chipsalliance/chisel #1632: Cat() not returning a literal for literal inputs

nrother, 2020-10-22:
When `Cat()` (or `##`) is called on literal nodes, the result is not a literal node anymore:

```scala
val hwInfo = Cat(numCores.U(4.W), buildNum.U(4.W))
assert(hwInfo.isLit()) //fails
```

This is at least surprising to a Chisel newbie like me. I known that the Cat function merely generated the corresponding FIRRTL statement, and constant folding is done later in the process, so this might be out-of-scope. On the other hand, printing out the value of literals is very handy for the automatic creating of documentation (register maps in my case).

One might consider adding a `LitCat()` function, which could than be even called outside of a Module context, which would be very handy for testing.

**Type of issue**: feature request

**Impact**: no functional change

**Development Phase**: request

**Please tell us about your environment:**

- version: `3.3.2`

**What is the use case for changing the behavior?**
Automatic generation of documentation, building literals in test code (in the `LitCat` case)

aswaterman, 2020-10-22:
Incidentally, there is an effort to support more constant-propagation in the frontend (for different reasons--to improve the performance and memory consumption of Chisel and Firrtl on large designs).  I think it would be cleaner to constant-propagate `##`/`Cat` as part of that effort, rather than introduce a new operator.

FWIW, I can understand why you find this behavior surprising, but normally I'd assemble these literals using Scala Ints or BigInts before constructing a UInt, e.g., `(numCores | buildNum << 4).U`.

nrother, 2020-10-26:
Ah, I'd love to see a more general overhaul of constant propagation, Chisel really feels a bit weak here (esp. in the realm of testing). Is there already an issue for tracking that?

This can be closed then, I guess.