chipsalliance/chisel #1648: Switch statements should not point to "Conditional.scala" in generated verilog

danielkasza, 2020-11-01:
**Type of issue**:  feature request

**Impact**: no functional change

**Development Phase**: request 

**Other information**
 

Example: https://scastie.scala-lang.org/vXpOaKjuQmmj91AUXLYGHQ

```verilog
module Foo(
  input        clock,
  input        reset,
  input  [1:0] io_in,
  output [1:0] io_out
);
  wire  _T = 2'h0 == io_in; // @[Conditional.scala 37:30]
  wire  _T_1 = 2'h1 == io_in; // @[Conditional.scala 37:30]
  wire  _T_2 = 2'h2 == io_in; // @[Conditional.scala 37:30]
  wire  _T_3 = 2'h3 == io_in; // @[Conditional.scala 37:30]
  wire [1:0] _GEN_0 = _T_3 ? 2'h3 : 2'h0; // @[Conditional.scala 39:67]
  wire [1:0] _GEN_1 = _T_2 ? 2'h1 : _GEN_0; // @[Conditional.scala 39:67]
  wire [1:0] _GEN_2 = _T_1 ? 2'h0 : _GEN_1; // @[Conditional.scala 39:67]
  assign io_out = _T ? 2'h3 : _GEN_2; // @[main.scala 13:10 main.scala 16:14 main.scala 20:14 main.scala 24:14 main.scala 28:14]
endmodule
```

**What is the current behavior?**

Verilog lines generated from switch statement refer back to `Conditional.scala`.

**What is the expected behavior?**

These lines should instead refer back to the source line of the `is` statement..

**What is the use case for changing the behavior?**

It would make it easier to cross-reference the generated Verilog and the Scala source.
I ran into this because one of my modules fails formal verification due to unreachable cases in a switch.

jackkoenig, 2020-11-01:
Thank you for the bug report! I believe this was fixed by #1618 (to be released in 3.4.1), but will verify before closing this issue.

danielkasza, 2020-11-12:
I just tried this on e6192ea75ce0d840b4b51a376921c2feecaa3b46 and I am still getting similar output.

jackkoenig, 2020-11-12:
Ah yes, I misread the issue originally. This is a limitation that needs to be addressed, thank you for checking!

PR #1669 (2020-11-16): Improve source locators for switch statements.
This fixes #1648.

### Contributor Checklist

- [ ] N/A ~~Did you add Scaladoc to every public function/method?~~
- [x] Did you add at least one test demonstrating the PR?
- [x] Did you delete any extraneous printlns/debugging code?
- [x] Did you specify the type of improvement?
- [ ] N/A ~~Did you add appropriate documentation in `docs/src`?~~
- [x] Did you state the API impact?
- [x] Did you specify the code generation impact?
- [x] Did you request a desired merge strategy?
- [ ] N/A ~~Did you add text to be included in the Release Notes for thi [... cut]
closed 2020-11-16 by seldridge