chipsalliance/chisel #1671: "operand is not visible from the current module" when trying to use BundleLiteral with RegInit

nrother, 2020-11-19:
**Type of issue**: bug report

I was trying to use the (experimental) BundleLiterals to specify the initial value of a Register, [as suggested on StackOverflow](https://stackoverflow.com/a/59055878). This produced a `chisel3.internal.ChiselException: "operand is not visible from the current module"`.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

```scala
import chisel3._
import chisel3.experimental.BundleLiterals._
import chisel3.stage.ChiselStage

class A extends Bundle {
  val a = Output(Bool())
}

class Example extends Module {
  val io = IO(new Bundle {
  })
  
  val reg = RegInit(new A().Lit(_.a -> false.B))
}
```
[Scastie Link](https://scastie.scala-lang.org/N2A9hSVSRWSjrwNpAHCToQ)

**What is the current behavior?**
```
chisel3.internal.ChiselException: operand is not visible from the current module
	at chisel3.internal.throwException$.apply(Error.scala:85)
	at chisel3.Data.requireVisible(Data.scala:455)
	at chisel3.Data.ref(Data.scala:480)
	at chisel3.RegInit$.apply(Reg.scala:179)
	at chisel3.RegInit$.apply(Reg.scala:192)
	at Playground$Example.<init>(main.scala:16)
	at Playground$$anonfun$2.apply(main.scala:20)
	at Playground$$anonfun$2.apply(main.scala:20)
	at chisel3.Module$.do_apply(Module.scala:54)
[...]
```

**What is the expected behavior?**
A correctly initialized register.

**Please tell us about your environment:**

- version: `3.4.0`

ducky64, 2020-11-30:
Preliminary diagnostic:
- When RegInit is called, it needs to get the `.ref` of the initializer
- In a Bundle / Aggregate type, the `.ref` is handled by `Data.ref`, which does the `requireVisible()` check
- But in an Element (eg, UInt) literal type, `ref` is overridden by `Element.ref`, which returns without calling the superclass ref if it's any kind of literal (either top-level literal, or an literal element in a Bundle literal)
- It's likely that `Data.requireVisible` would fail on any literal (if it were not short circuited by `Element.ref`, since it checks the `location` of all `TopBindings` (including `LitBinding`, which is a subclass of `UnconstrainedBinding`, which has `location` defined to be `None`, because it's meant to be not bound anywhere)
  - A quick solution to get around this would be for `requireVisible` to have a `case Some(ub: UnconstrainedBinding) =>`
- But BundleLiterals still don't have a valid ref (it crashes because it can't handle literals - wow, surprised this hasn't come up before), because that never gets invoked on the BundleLiteral as a whole when involved in a connect operation - instead, MonoConnect / BiConnect breaks it into element-wise operations.

Potential solutions:
- From a clean code viewpoint, having `Element.ref` short circuit the `Data.ref` checks is probably non-ideal. Perhaps `Data` should define a function to be overridden in the literal case, and maybe `Data.ref` should be `final`.
- FIRRTL support for Bundle Literal types would give it a node akin to `Element.ref` and is probably the clean solution. I'm not sure what the quick-n-dirty Chisel solution is.
- I'm not sure there's a great user-code workaround, aside from handling the reset explicitly in user-code with assigns, or breaking the bundle into individual registers. Register init is otherwise a FIRRTL primitive construct.

jackkoenig, 2020-11-30:
Thanks for the diagnosis work @ducky64. FIRRTL support would make it cleaner, but I think we can and should this without that larger improvement. On the point of `Element.ref` short circuiting `Data.ref`, I bet that is just historical since previously, only `Elements` could be literals. I agree with you that we should just make `Data.ref` `final` and handle all literals there so that the support for Aggregates is in the same place as for `Elements`.

jackkoenig, 2020-11-30:
There is actually a fairly easy user-code workaround, just wrap the Bundle-lit in `WireInit`: https://scastie.scala-lang.org/f656eVYVTgqE9e3iwuegWQ

```scala
  val reg = RegInit(WireInit(new A().Lit(_.a -> false.B)))
```

chick, 2020-11-30:
Or an ugly internal hack
```scala
    reg.bind(RegBinding(Builder.forcedUserModule, Builder.currentWhen()))
    requireIsHardware(init, "reg initializer")

    try {
      pushCommand(DefRegInit(sourceInfo, reg, clock.ref, reset.ref, init.ref))
    }
    catch {
      case e: ChiselException if e.getMessage.contains("operand is not visible from the current module") =>
        val wrapped = WireInit(init)
        pushCommand(DefRegInit(sourceInfo, reg, clock.ref, reset.ref, wrapped.ref))
    }
```
seems to work

ducky64, 2020-11-30:
Given the above, I think the less-hacky-but-still-hacky solution would be to fix requireVisible properly, then override Bundle.ref to handle the literal case with the above wrapper and a TODO for proper bundle literal support?

jackkoenig, 2020-12-01:
Related bug in Printable: https://scastie.scala-lang.org/trNY7r6fTLSTVhWa3MalnQ

PR #1688 (2020-12-02): Fix RegInit of Bundle lits
Implemented by folding Element.ref into Data.ref. Element.ref had
special handling for literals, but because Bundles can also be literals,
there were code paths that tried to get the ref of a Bundle literal
which was non-existent. Now, all literals are handled together.

Because FIRRTL does not have support for Bundle literals, Bundle literal
refs are implemented by materializing a Wire.

This also handles the DontCare fields of partially initialized Bundle literals in the same way.

Fixes #1671

I could not find a good way to fix the related issue with Bundle litera [... cut]
closed 2020-12-02 by jackkoenig