chipsalliance/chisel #1685: RefNotInitializedException when use switch

lxu28973, 2020-11-30:
**Type of issue**: bug report

when I write MUX using switch block,   one exception be thrown:
[info]    firrtl.passes.CheckInitialization$RefNotInitializedException: : [module Mux4]  Reference io is not fully initialized.
[info]    @[Mux4.scala 43:12 Mux4.scala 46:12 Mux4.scala 49:12 Mux4.scala 52:12] : io.y <= mux(_T, io.a, mux(_T_1, io.b, mux(_T_2, io.c, mux(_T_3, io.d, VOID)))) @[Mux4.scala 43:12 Mux4.scala 46:12 Mux4.scala 49:12 Mux4.scala 52:12]

But I think there is no wire not connect, why it throw RefNotInitializedException, also why chisel try to use so much MUXs?
Is it a bug of chisel?

Here is my code:
```
class Mux4 extends Module {
  val io = IO(new Bundle {
    val a = Input(UInt(1.W))
    val b = Input(UInt(1.W))
    val c = Input(UInt(1.W))
    val d = Input(UInt(1.W))
    val sel = Input(UInt(2.W))
    val y = Output(UInt(1.W))
  })
  switch(io.sel) {
    is("b00".U) {
      io.y := io.a
    }
    is("b01".U) {
      io.y := io.b
    }
    is("b10".U) {
      io.y := io.c
    }
    is("b11".U) {
      io.y := io.d
    }
  }
}
```
Chisel version I use is 3.3.1

ekiwi, 2020-11-30:
This is a known issue. The compiler is not smart enough to know that your choices are exhaustive.
I think for comparison to constant it would be nice to add a special case to the compiler to detect exhaustiveness. Maybe we can keep this issue open for feature tracking.

Lyon-ai, 2021-07-14:
Hello, and there is another problem, when I use the code above and then run **sbt test:runMain modules.Launcher chiselPra** order in terminal, it reports an error as followed:
polymorphic expression cannot be instantiated to expected type;
[error]  found   : Set         (in scala.collection)
[error]  required: Set[BigInt] (in scala.collection.immutable)
[error]   switch(io.sel) {
[error]                  ^
[error] one error found
[error] (Compile / compileIncremental) Compilation failed

But this error doesn't appear when I run it using the jupyter notebook with scala Kernel.
Why does this error happen and what should I do to fix this error?
my sbt version is 1.4.9