chipsalliance/chisel #1712: RegInit(VecInit vs VecInit( ... RegInit

mysoreanoop, 2020-12-23:
**Type of issue**: bug report | feature request | documentation | other enhancement
other enhancement

**Impact**: no functional change | API addition (no impact on existing code) | API modification | unknown
no functional change

**Development Phase**: request | proposal
request(?)
**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
Using, for example, 
val x = VecInit(Seq.fill(4) (RegInit(0.U(10.W))))
and then later updating like so:
x(2) := x(2) + 4.U
throws a combinational loop error pointing to this line.
However, having val x = RegInit(VecInit(Seq.fill(4) (0.U(10.W)))) doesn't have that problem.
**What is the current behavior?**
Throws a combinational loop error
**What is the expected behavior?**
FIRRTL/some other entity should identify that it's a syntax/style error, perhaps? Or at least suggest using an alternative like RegInit(VecInit instead of VecInit(...RegInit like with other cases
**Please tell us about your environment:**
 - version: 3.3-SNAPSHOT
 - OS: Linux oxygen 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
**What is the use case for changing the behavior?**
Would save a lot of time in figuring out if it really is a combinational loop error as not everybody can go through the exhaustive process of familiarising themselves with the coding style/syntax thoroughly.

ekiwi, 2020-12-23:
This seems like a read usability issue.

One question that I would be interested to know the answer to: Why did you decide to wrap `Seq.fill(4) (RegInit(0.U(10.W)))` in `VecInit`? If you do not need dynamic (i.e. at hardware runtime) indexing, then just using a `Seq` which can be indexed at elaboration (i.e. Scala runtime) should be sufficient.

schoeberl, 2020-12-23:
The error message is correct, because you described a combinational loop. Luckily this is not possible in Chisel (it is in VHDL, Verilog, and SystemVerilog, where you sometimes not even get a warning).

Your code is similar to:

val s = 0.U
s := s + 4.U

You cannot have a signal on the left hand side and on the right hand side (in concurrent hardware). A register in between breaks the combinational loop.

Cheers,
Martin

PS: tried to find a good article (or Wikipedia) on the topic, but did not find one. Maybe we should write one ;-)

ekiwi, 2020-12-24:
> You cannot have a signal on the left hand side and on the right hand side (in concurrent hardware). A register in between breaks the combinational loop.

I assume that's what the OP was trying to do. `x` is supposed to be a vector register (or a list of registers).

mysoreanoop, 2020-12-24:
@ekiwi The example was statically indexed, but I indeed needed a dynamically indexable vector of registers for updating individual entities of the vector conditionally.

@schoeberl Contrasting your example with my example code, val s is set up as a register. And the initial value is only for when the reset is lifted, right?

schoeberl, 2020-12-29:
Yes, using a register is fine.