chipsalliance/chisel #1715: emitVerilog on the ChiselStage object

schoeberl, 2021-01-05:
**Type of issue**: feature request

**Impact**: API addition (no impact on existing code)

```emitVerilog``` on the ChiselStage companion object should generate a Verilog file (like the class function).

Furthermore, having the args in the object method would be nice too, as one most of the time wants to specify the target folder.

jackkoenig, 2021-05-03:
We did this intentionally but I totally agree that the inconsistency is confusing. When we switched things from `Driver` to `ChiselStage` we realized we didn't have any way of running Chisel and FIRRTL without writing files so we wanted an API to not write files. That being said, this wasn't the right way to do it. 

So we should make them consistent as this issue says, and remember that the default case is writing files. `ChiselStage` being the name of the main entry point is a bit confusing on its own, perhaps we should add an object `Chisel3`?

schoeberl, 2021-05-05:
I would propose to have then two methods: ```emitVerilog()``` to write a file (returning nothing, i.e., ```Unit```) and ```getVerilog()``` returning just the string. Where to put it? Maybe on an object ```Chisel3```? However, this is a method that does something with a (new) module. So logically it should be a method on ```Module```.

schoeberl, 2021-05-18:
I am trying to implement this. However, I am a bit confused about the splitting of Chisel code into ```src/main/...```, which contains utilities and stage stuff and code that lives in ```core/src/main/...```. I am not able, or don't know, how I can access code from src/main in core/src/main. The compiler simply tells me that chisel3.stage does not exist.

What is the reason for splitting the Chisel source into two folders? And why is the access only possible from one direction into the other?

Sorry for spamming this thread with basic Chisel development questions.

jackkoenig, 2021-05-18:
> Sorry for spamming this thread with basic Chisel development questions.

No worries at all, this should be in a `CONTRIBUTORS` doc but alas.

> What is the reason for splitting the Chisel source into two folders? And why is the access only possible from one direction into the other?

There are actually 4 projects in Chisel, as separate compilation units.
* `plugin` is the compiler plugin, no internal dependencies
* `macros` is most of the macros used in Chisel, no internal dependencies
* `core` is the bulk of the source code of Chisel, depends on `macros`
* `src/main` is sort of the "main" that brings it all together and includes `util`, depends on `core`

Other than the plugin, the reason there are 3 is a fundamental limitation in Scala that you cannot define and use macros in the same compilation unit. `macros` obviously implements some macros, but so does `core` (see [CompileOptions](https://github.com/chipsalliance/chisel3/blob/83eeda2efebb1e3e3d66365d388f5de627766d47/core/src/main/scala/chisel3/CompileOptions.scala)).

One may wonder why the `CompileOptions` macros live in `core` instead of `macros`. In this case, we're taking advantage of the previously mentioned limitation--lots of Chisel APIs take an implicit CompileOptions (which is how we implement `import chisel3._` vs. `import Chisel._`), the macro will materialize a default if none is provided. We found it very error prone to ensure we're propagating the user-provided CompileOptions rather than accidentally materializing them via the macro. By moving the macro implementation into `core`, we **cannot** screw this up any more; you'll get a compiler error if you don't propagate `CompileOptions` properly.

While it's not super clean cut, the basic dividing line between what belongs in `core` vs. `src/main` is that things that touch lots of APIs that are private to package chisel3 tend to live in `core`, while stuff that is either pure chisel (eg. `util`) or things that are mainly just code on top (like Stage) live in `src/main`.

schoeberl, 2021-05-21:
Could we move ```stage.*``` into ```core```? In that case we could simple call the relevant functions from ```RawModule```.

PR #1921 (2021-06-16): Verilog
### Contributor Checklist

This is just a start for a discussion to resolve  #1715. Not intended to merge now, before we finished the discussion.

This is a quick fix to add Verilog emission to a Chisel3 object, instead of ChiselStage or new ChiselStage. However, I would prefer to have this functionality on ```Module```, like this:

```
(new Hello()).emitVerilog
```

where ```Hello()``` is a Chisel Module.
closed 2021-06-16 by mergify[bot]