chipsalliance/chisel #1722: only last register of ShiftRegister can be renamed by suggestName

sequencer, 2021-01-12:
**Type of issue**: bug report 
**Impact**: no functional change

**Development Phase**: request

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
https://scastie.scala-lang.org/BUm3SiiCTA6DRLi9b51YPQ

**What is the current behavior?**
`suggestName` only rename last register.

**What is the expected behavior?**
`suggestName` should rename all register generated by `ShiftRegister`

mwachs5, 2021-01-12:
https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/util/ShiftReg.scala#L10

sequencer, 2021-01-12:
Thank @mwachs5, I saw this, but it needs user providing a optional name, rather than using `suggestName` API.
And another problem is if user wanna annotate internal register, current implementation is not possible to expose those register to users.
So I add new API `ShiftRegisters` to expose each register if needed.
More example will be added later.

sequencer, 2021-01-12:
For example, usage like this is really helpful to users :p

https://github.com/chipsalliance/chisel3/blob/07c9fd234c40635b25b4d1c3db48527d2d17973b/src/main/scala/chisel3/util/experimental/crossing/AsyncQueue.scala#L168-L172

unlike constructing a new `Module` explicitly in rocket-chip, it just use `chisel3.util.experimental.group` API to annotate all register in a ShiftRegisters, gather them together in a new Module.