chipsalliance/chisel #1726: Some Modules with different IOs

escou64, 2021-01-13:
**Type of issue**: bug report

**Impact**: unknown

**Informations**:
I have two modules with different IOs:
```
class Mod0 extends Module {
  val io = IO(new Bundle {
    val in = Input(UInt(32.W))

    val out = Output(UInt(32.W))
  })

  io.out := io.in + 44.U
}
```

```
class Mod1 extends Module {
  val io = IO(new Bundle {
    val in = Input(UInt(32.W))
    val in2 = Input(UInt(32.W))

    val out = Output(UInt(32.W))
  })

  io.out := io.in + io.in2
}
```
In another module, I want to create an instance of one module depending a condition.
I use only one variable because most of the IOs will be connected to the same wires, but some IOs are changing:
```
val mod = if (Condition) Some(Module(new Mod1())) else Some(Module(new Mod0()))
mod.get.io.in := 32.U
if (Condition) mod.get.io.in2 := 64.U
```

**What is the current behavior?**
An error occurs during compilation:
```
value in2 is not a member of chisel3.Bundle{val in: chisel3.UInt; val out: chisel3.UInt}
```
It seems that both modules are checked every time while only one is really used at any given time.

**What is the expected behavior?**

I expected that the IOs could change in the same way as the internal code.
Maybe this is just a bad way to do it?

ekiwi, 2021-01-13:
The problem that you are encountering is the strong static typing in Scala.
One way to work around this could be:
```scala
val modOut = if (Condition) {
    val mod = Module(new Mod1())
    mod.io.in := 32.U
    mod.io.in2 := 64.U
    mod.io.out
  } else {
    val mod = Module(new Mod0())
    mod.io.in := 32.U
    mod.io.out
  }
```

or

```scala
val mod = if (Condition) Some(Module(new Mod1())) else Some(Module(new Mod0()))
mod.get.io.in := 32.U
mod match {
  case Some(m: Mod1) => m.io.in2 := 64.U
  case _ =>
}
```

(I haven't compiled these, so there could be typos or other problems)

mwachs5, 2021-01-13:
Since `Mod0` and `Mod1` don't share any common superclass or trait other than `Module`, that is what the type of `mod` is inferred to be. So, the scala type system doesn't know anything about `mod.io` other than what it would know about `Module`

closed 2021-03-11 by ekiwi