chipsalliance/chisel #1729: Expose AnnotationSeq to Builder

sequencer, 2021-01-14:
**Type of issue**: feature request

**Impact**: API addition (no impact on existing code)

**Development Phase**: proposal

**What is the use case for changing the behavior?**
Current situation is that after changing a small part of codes, whole RTL will be re-elaborated.
But for time consuming part like a very large decoder, which won't need the QMC re-computation if don't change the decoder table.
So I propose a simple idea that, we can expose `AnnotationSeq` to `Builder`, served as a standard interface to user, to cache decoder table during different elaborations.

ekiwi, 2021-01-14:
I like the idea of having some way of caching results, however, I do not understand how `AnnotationSeq` would help with that. Do you have an example on how this would work?

sequencer, 2021-01-14:
Just serialize decode table, put it in a file, use command-line read it back to `caches`.
Read annotation from here:
https://github.com/chipsalliance/chisel3/blob/e9612f6bc3bb707de89fc25e9672419dafccd433/src/main/scala/chisel3/util/experimental/decoder/QMCDecoder.scala#L12
Dump decode table to a Annotation from here:
https://github.com/chipsalliance/chisel3/blob/e9612f6bc3bb707de89fc25e9672419dafccd433/src/main/scala/chisel3/util/experimental/decoder/QMCDecoder.scala#L161

ekiwi, 2021-01-14:
Have you considered using something like `ScalaCache` instead? https://cb372.github.io/scalacache/docs/
We probably would want to write a wrapper to properly namespace keys (maybe also tag them with the chisel/firrtl version in order to invalidate them when those change) and add [a simple filesystem backend](https://gist.github.com/Grogs/f8a53f3757951293cf978a22a178aa99) so that people do not need to run a reddis server if they do not want to.

We do not actually need to use this library, but it might serve as an interesting starting point for a caching API that could maybe later be reused for incremental builds or similar.

sequencer, 2021-01-14:
https://github.com/chipsalliance/chisel3/commit/ec91e57e550a05603187bcc92210b32f37b51b24 is the example, but serialization and deserialization is is not implemented.
I think it's a good example to illustrate this idea.

> Have you considered using something like ScalaCache instead?

As for caching, I think decode table is easy to cache, it's a deterministic key->value Map. It's safe to store them during different elaboration.

But if we come to talk about caching circuit, It's a more complicated build-system problem:
It's easy to dump everything to cache, but not easy to determine which part should be re-evaluated, If a `Module` depends on a global value to elaborate, even more, if a Module depends on timestamp, how to do we be able to find the cache-able part?

So my idea is let user add their own Annotation to cache, rather than facing build-system problem directly, user can add a FIRRTL blackbox on their own with Annotations.

PR #1731 (2021-02-26): Expose AnnotationSeq to Module.
Resolve #1729
depends on #1730
### Contributor Checklist

- [x] Did you add Scaladoc to every public function/method?
- [x] Did you add at least one test demonstrating the PR?
- [x] Did you delete any extraneous printlns/debugging code?
- [x] Did you specify the type of improvement?
- [ ] Did you add appropriate documentation in `docs/src`?
- [x] Did you state the API impact?
- [x] Did you specify the code generation impact?
- [x] Did you request a desired merge strategy?
- [x] Did you add text to be included in the Release Notes for this change?

#### Type of I [... cut]
closed 2021-02-26 by jackkoenig