chipsalliance/chisel #1734: how declare dff without reset ??

CaseyZhu, 2021-01-15:
**Type of issue**: bug report | feature request | documentation | other enhancement

**Impact**: no functional change | API addition (no impact on existing code) | API modification | unknown

**Development Phase**: request | proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

CaseyZhu, 2021-01-15:
Type of issue: feature request

sequencer, 2021-01-15:
https://scastie.scala-lang.org/t0tsPfXGSFmJAmiOYc1kkA

closed 2021-01-15 by sequencer

schoeberl, 2021-01-15:
Why not just:

RegNext(input)