chipsalliance/chisel #1764: "when" is not working

lxu28973, 2021-01-29:
**Type of issue**: bug report

**Impact**:  unknown

**Other information**
 
I use the chipyard v1.3.0, and change some code in the BOOM generator.
As usual, the when condition will check the default value. But I find that one of my when blocks didn't work. I havn't assigned the default value, but it didnot report a error like "Reference io is not fully initialized". My code: 
```
for (i <- 0 until numReadPort){
    when(delogic.io.zeroDetectOut(i).resZero){
      io.wp(i).iresp.bits.data := 0.U
      io.wp(i).iresp.valid := io.rp(i).req.valid && !IsKilledByBranch(io.rp(i).brupdate, io.rp(i).req.bits.uop)
      io.wp(i).iresp.bits.uop := io.rp(i).req.bits.uop
      io.wp(i).iresp.bits.uop.br_mask := GetNewBrMask(io.rp(i).brupdate, io.rp(i).req.bits.uop)
    }
  }
```
I see the verilog it generated, and find it just like "when(delogic.io.zeroDetectOut(i).resZero)" was not exist. It does not use any mux. For example: 
```
  assign io_wp_0_iresp_valid = io_rp_0_req_valid & _T_36; // @[MultiPortExeUnit.scala 84:28:chipyard.TestHarness.My6WideBoomConfig.fir@406003.6]
```
The _T_36 just equal to "!IsKilledByBranch(io.rp(i).brupdate, io.rp(i).req.bits.uop)" in Chisel.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
I try some other simple conditions, but they worked well. Maybe it only show up in big projects.

mwachs5, 2021-01-29:
is it possible that `delogic.io.zeroDetectOut(i).resZero` is always true?

jackkoenig, 2021-01-29:
What does your Chisel import look like? Is it `import chisel3._` or `import Chisel._`? If it is the latter (`import Chisel._`) all signals are invalidated by default.

> I havn't assigned the default value

Just to check, you didn't connect to `DontCare` or anything? chipyard v1.3.0 uses Chisel 3.3 which certainly usually catches not fully initialized ports (https://scastie.scala-lang.org/DT5Bw1iLTJm4xh3Q1yUZpg) so if you're using `import chisel3._` and truly don't have any other connections to your port, is it possible you could share your source code or at least a generated `.fir` or `.pb`?

lxu28973, 2021-01-30:
Thankyou for your reply. 
I try to modify my code to:
```
for (i <- 0 until numReadPort){
    when(delogic.io.zeroDetectOut(i).resZero){
      io.wp(i).iresp.bits.data := 0.U
      io.wp(i).iresp.valid := io.rp(i).req.valid && !IsKilledByBranch(io.rp(i).brupdate, io.rp(i).req.bits.uop)
      io.wp(i).iresp.bits.uop := io.rp(i).req.bits.uop
      io.wp(i).iresp.bits.uop.br_mask := GetNewBrMask(io.rp(i).brupdate, io.rp(i).req.bits.uop)
    }.otherwise{
      io.wp(i).iresp.valid := false.B
    }
  }
```
Now, I get the verilog: 
```
 assign io_wp_0_iresp_valid = delogic_io_zeroDetectOut_0_resZero & _T_37; // @[MultiPortExeUnit.scala 84:28:chipyard.TestHarness.My6WideBoomConfig.fir@406003.6 MultiPortExeUnit.scala 88:28:chipyard.TestHarness.My6WideBoomConfig.fir@406089.6]
```
where the _T_37 is 
```
  assign _T_37 = io_rp_0_req_valid & _T_36; // @[MultiPortExeUnit.scala 84:50:chipyard.TestHarness.My6WideBoomConfig.fir@406002.6]
```
Same as befer, the _T_36 just equal to "!IsKilledByBranch(io.rp(i).brupdate, io.rp(i).req.bits.uop)" in Chisel. It's now have the delogic_io_zeroDetectOut_0_resZero  in verilog, so I dont think that it is  because delogic.io.zeroDetectOut(i).resZero is always true.

Here is my import:
```
import scala.collection.mutable.ArrayBuffer
import chisel3._
import chisel3.util._
import freechips.rocketchip.config.Parameters
import freechips.rocketchip.rocket.{BP, PipelinedMultiplier}
import freechips.rocketchip.tile.{RoCCCoreIO, XLen}
import freechips.rocketchip.tile
import FUConstants._
import boom.common._
import boom.ifu.GetPCFromFtqIO
import boom.util.{BoomCoreStringPrefix, BranchKillableQueue, GetNewBrMask, ImmGen, IsKilledByBranch}
import freechips.rocketchip.util.UIntToAugmentedUInt
```

[sourceCode.zip](https://github.com/chipsalliance/chisel3/files/5897388/sourceCode.zip)
I upload my scala file and a generated .fir of the Top.
The code mentioned befer can be found in MultiPortExeUnit.scala, line 81 to 90.

jackkoenig, 2021-02-02:
The issue is that your module extends `MulExeUnit` which extends `MultiPortExeUnit` (also your module) which extends [`BoomModule`](https://github.com/riscv-boom/riscv-boom/blob/840d4e51ce0c4319f0ae8243487a137fd9745963/src/main/scala/common/types.scala) which extends rocket-chip's [`CoreModule`](https://github.com/chipsalliance/rocket-chip/blob/58a0e1052570e42caafd9987ed0146dd012c57df/src/main/scala/tile/Core.scala#L128) which extends `Module` but uses `import Chisel._`. Unfortunately, when using inheritance, it is the Module at the top of the inheritance tree that sets the behavior of whether or not signals are invalidated by default. 

This is unfortunately intended behavior, ideally either rocket-chip would stop doing `import Chisel._`, or other projects would not inherit from its Modules.

Thinking about this more, we probably should have made the stricter connection semantics bind on definition of the signal (eg. `val io = IO(...)`) rather than the definition of the module (which we can only do at the top of the inheritance hierarchy, ie. we can only associate information about which import is used where the `extends Module` code is located). This would be a hard thing to change as this point as it could potentially break a lot of otherwise working code, but it's worth considering.

lxu28973, 2021-02-02:
Yeah, this is the problem. Thank you veay much!

closed 2021-02-02 by lxu28973