chipsalliance/chisel #1784: Name sanitation behavior changed between Chisel v3.3 and v3.4

jackkoenig, 2021-02-10:
From https://github.com/chipsalliance/chisel3/pull/1624#issuecomment-711415026

**Type of issue**: bug report

**Impact**: no functional change | API addition (no impact on existing code) | API modification | unknown

**Development Phase**: request | proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

```scala
class MyBundle extends Bundle {
  val foo = UInt(8.W)
  val `foo😎` = UInt(8.W)
  val `foo😎😎` = UInt(8.W)
}

class Example extends MultiIOModule {
  val in = IO(Input(new MyBundle))
  val out = IO(Output(new MyBundle))
  
  out := in
}
```

**What is the current behavior?**

In v3.4.2 gives:
```verilog
module Example(
  input        clock,
  input        reset,
  input  [7:0] in_foo,
  output [7:0] out_foo
);
  assign out_foo = in_foo; // @[main.scala 16:7]
endmodule
```
https://scastie.scala-lang.org/TcWA1dnGSEyfNeqOSM9i5g

**What is the expected behavior?**

In v3.3.3 this gives
```verilog
module Example(
  input        clock,
  input        reset,
  input  [7:0] in_foo,
  input  [7:0] in_foouD83DuDE0E,
  input  [7:0] in_foouD83DuDE0EuD83DuDE0E,
  output [7:0] out_foo,
  output [7:0] out_foouD83DuDE0E,
  output [7:0] out_foouD83DuDE0EuD83DuDE0E
);
  assign out_foo = in_foo; // @[main.scala 16:7]
  assign out_foouD83DuDE0E = in_foouD83DuDE0E; // @[main.scala 16:7]
  assign out_foouD83DuDE0EuD83DuDE0E = in_foouD83DuDE0EuD83DuDE0E; // @[main.scala 16:7]
endmodule
```
https://scastie.scala-lang.org/f43qpaw6SYCaA2Gg9mXPXg

I'm not going to argue those names are better, in fact, perhaps `v3.4.2` is better, but the fact that the ports collide down to one is problematic.

**What is the use case for changing the behavior?**

Don't have ports disappear randomly and gracefully handle unicode names.

mwachs5, 2021-02-10:
"Don't have ports disappear randomly and gracefully handle unicode names." -- is erroring on unicode names acceptable/graceful?

johnsbrew, 2021-02-10:
> "Don't have ports disappear randomly and gracefully handle unicode names." -- is erroring on unicode names acceptable/graceful?

I would go for that since unicode characters are known to be controversial in several cases, so better not to use them at all 
- https://nrinaudo.github.io/scala-best-practices/tricky_behaviours/unicode_operators.html
- https://github.com/scala/scala/pull/7540

jackkoenig, 2021-02-10:
I think erroring is reasonable and probably desirable.