chipsalliance/chisel #1787: RegInit errors when passed a Bundle Literal

jackkoenig, 2021-02-18:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: request

**Other information**
 

Using Chisel v3.4.2

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

https://scastie.scala-lang.org/4CDkXkNHQpKV7ItnPRtcvw

```scala
import chisel3._
import chisel3.experimental.BundleLiterals._

class MyBundle extends Bundle {
  val a = UInt(8.W)
  val b = Bool()
}

class Example extends MultiIOModule {
  val in = IO(Input(new MyBundle))
  val en = IO(Input(Bool()))
  val out = IO(Output(new MyBundle))
  
  // It works if you wrap the Bundle lit in WireInit
  val r = RegInit((new MyBundle).Lit(_.b -> true.B))
  when (en) {
    r := in
  }
  out := r
}
```

**What is the current behavior?**

```
chisel3.internal.ChiselException: operand is not visible from the current module
```

**What is the expected behavior?**

This should work, and does if you wrap the Bundle Literal in `WireInit`

**What is the use case for changing the behavior?**

Fix a bug

PAN-Ziyue, 2021-04-11:
I met the same issue. Is there any chance to fix that bug?

jackkoenig, 2021-04-12:
> I met the same issue. Is there any chance to fix that bug?

It's still not fixed, but the workaround is pretty simple, you can wrap the Bundle literal in `WireInit(...)`, eg.
```scala
val r = RegInit(WireInit((new MyBundle).Lit(_.b -> true.B)))
```
https://scastie.scala-lang.org/jackkoenig/eXjAqmDRRfOMYCMqNkRNXQ/1

jackkoenig, 2021-05-03:
Maybe fixed by https://github.com/chipsalliance/chisel3/pull/1689