chipsalliance/chisel #1808: RegInit initial value does not take effect

itviewer, 2021-03-12:
**Type of issue**: bug report 

**What is the current behavior?**
Declare register with initial value
```
  val x = RegInit(1.U(16))
  val y = RegInit(2.U(16))
```
Produces the following FIRRTL and Verilog
```
    reg x : UInt<1>, clock with : (reset => (reset, UInt<1>("h00"))) @[GCD.scala 20:18]
    reg y : UInt<1>, clock with : (reset => (reset, UInt<1>("h00"))) @[GCD.scala 21:18]
```
```
  always @(posedge clock) begin
    if (reset) begin // @[GCD.scala 20:18]
      x <= 1'h0; // @[GCD.scala 20:18]
    end else begin
      x <= _GEN_2[0];
    end
    if (reset) begin // @[GCD.scala 21:18]
      y <= 1'h0; // @[GCD.scala 21:18]
    end else begin
      y <= _GEN_3[0];
    end
  end
```
**What is the expected behavior?**
```
    reg x : UInt<1>, clock with : (reset => (reset, UInt<1>("h01"))) @[GCD.scala 20:18]
    reg y : UInt<1>, clock with : (reset => (reset, UInt<1>("h02"))) @[GCD.scala 21:18]
```
**Please tell us about your environment:**
Chisel 3.5-SNAPSHOT and 3.4.2

itviewer, 2021-03-12:
Sorry, my mistake, I lost W , but it shouldn’t be compiled in this way.

closed 2021-03-12 by itviewer

edwardcwang, 2021-03-12:
#1184 strikes again 😉

jackkoenig, 2022-06-02:
Finally fixed in https://github.com/chipsalliance/chisel3/pull/2534 🙂