chipsalliance/chisel #1812: Feature Request: Ensure unique filenames in FileInfo

ekiwi, 2021-03-16:
**Type of issue**: feature request

**Development Phase**: request

**Other information**

Currently the `FileInfo` that we produce only includes the name of the scala file without any path info. This is great if there is only a single file with this name. However, once there is more than one file with the same name in the code base, it is no longer possible to automatically jump to the correct scala file.

As an example, let's say we have the following files:
- `src/core/Alu.scala`
- `src/accelerator/Alu.scala`
- `src/accelerator/TensorAlu.scala`

Currently we would get `Alu.scala` for lines from both files of this name.

I would like to suggest, that we selectively add path information until we get a unique name for each file.
So in this case we would use the following names in the `FileInfo`: `core/Alu.scala`, `accelerator/Alu.scala`, `TensorAlu.scala`.

This way of distinguishing filenames is similar to what IntelliJ does:

[image]

I think this would be a nice compromise which would help us distinguish files with the same name, but still retain the human readable `FileInfo` with minimal changes.

ekiwi, 2021-03-16:
Problem: libraries that contain source locators.

sequencer, 2021-03-22:
related to #1517
main issue is Chisel don't don't where is the root of project, which is a metadata from build system.
But if split by `/` and a a Suffix Tree might solve this issue

itviewer, 2021-03-25:
@ekiwi 
The following situations also need to be considered
```
scala/core/Alu.scala
test/scala/core/Alu.scala
```
I suggest to provide a file list for the user to choose which file to jump to. If you use [easysoc-firrtl](https://github.com/easysoc/easysoc-firrtl) plugin, I have added this feature and uploaded the new version to the Intellij plugin store. (It may take a day for the update to be available, before the plugin is approved, you can manually [download](https://plugins.jetbrains.com/plugin/download?rel=true&updateId=114795) and install it)

[image]

ekiwi, 2021-03-25:
> I suggest to provide a file list for the user to choose which file to jump to. If you use [easysoc-firrtl](https://github.com/easysoc/easysoc-firrtl) plugin, I have added this feature and uploaded [...]

This file list is really cool for the GUI situation. What I am currently looking at is generating a coverage report, which is supposed to work without user interaction.

Btw. if there is any way that we could integrate the coverage information into your firrtl plugin that would be super cool. If you want to checkout the current prototype, you can find it here: https://github.com/ucb-bar/chisel-testers2/pull/274

ekiwi, 2023-01-12:
fixed by https://github.com/chipsalliance/chisel3/pull/2791

closed 2023-01-12 by ekiwi