chipsalliance/chisel #1829: formatting issue in the "Experimental Features" part of the website

mwachs5, 2021-03-22:
**Type of issue**: documentation

**Impact**: no functional change

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

Go to https://www.chisel-lang.org/chisel3/docs/appendix/experimental-features.html

**What is the current behavior?**

It looks like this:

[image]

**What is the expected behavior?**

There should be a bulleted list of links not all one line

**Please tell us about your environment:**

Chrome

**What is the use case for changing the behavior?**

Make it easier to navigate the docs/website

klmrdn, 2021-03-31:
I tried to fix it and made a pull request #1844

jackkoenig, 2021-05-03:
This is fixed.

closed 2021-05-03 by jackkoenig