chipsalliance/chisel #1831: The Wiki and docs in the repository don't match

KasperHesse, 2021-03-23:
**Type of issue**: documentation

**Impact**: no functional change

**Development Phase**: request

**Other information**
 
The wiki located at https://github.com/chipsalliance/chisel3/wiki seems outdated compared to the documentation found at https://github.com/chipsalliance/chisel3/tree/master/docs/src. The wiki pages mainly seem to be from mid-2019, whereas documentation pages are regularly updated. I would expect it to be relatively simple to migrate the wiki to use the documentation in src/docs. I would recommend that the wiki pages be updated to reflect the information stored in the docs.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
N/A

**What is the current behavior?**
N/A

**What is the expected behavior?**
N/A

**Please tell us about your environment:**
N/A

**What is the use case for changing the behavior?**
It will be easier for users and developers to view up-to-date information when it is presented in the wiki. Additionally, having two different resources (with sometimes contradicting information) doesn't seem like the desired functionality.

mwachs5, 2021-03-23:
I believe the plan @jackkoenig? is to just delete the wiki

azidar, 2021-03-23:
I think deleting the wiki is correct, perhaps just add a landing page to the website?

jackkoenig, 2021-03-23:
Thank you @KasperHesse for pointing this out. The docs in the repo are now hosted in the chisel3 section of chisel-lang.org, ie. https://www.chisel-lang.org/chisel3

Now that all of the content from the Wiki is properly migrated, I think it's time to retire it. A link makes the most sense.

jackkoenig, 2021-03-23:
Okay I changed the landing page of the Wiki to link to the website and deleted the sidebar. Hopefully more clear now.

closed 2021-03-23 by jackkoenig

ucbjrl, 2021-03-23:
Someone should ensure that the Jenkins jobs that tried to keep the public and private versions of the wiki in sync is disabled. I think they're the only Jenkins jobs still active.

jackkoenig, 2021-03-23:
Thanks for the heads up @ucbjrl! Is this something you can take a look at @chick?

chick, 2021-03-23:
Putting this on my list