chipsalliance/chisel #1846: Verilog attribute annotations in modules are being lost when dedupping

Quarky93, 2021-04-01:
**Type of issue**: bug report

**Other information**

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
Here is an example where a Verilog attribute is being emitted correctly if only one module is instantiated and not emitted when there are two duplicate modules.
https://scastie.scala-lang.org/Quarky93/SIjXX7lgRj2Q3iS19PA0Cg/22

**What is the current behavior?**
Verilog attribute is emitted correctly if only one module is instantiated and not emitted when there are duplicate modules after deduplication.

**What is the expected behavior?**
Attribute annotations should be kept after deduplication.

**Please tell us about your environment:**
chisel 3.4.2

jackkoenig, 2021-07-14:
Fixed by https://github.com/chipsalliance/firrtl/pull/2286, to be released in FIRRTL 1.4.4 (Chisel 3.4.4)

jackkoenig, 2021-07-20:
chipsalliance/firrtl#2286 fixed the issue for memory annotations but not attributes.

Quarky93, 2022-01-10:
Is there a planned fix for attributes?

carlosedp, 2022-01-10:
I think there is still some pending point on memory dedup with initialization annos like described here:
https://github.com/chipsalliance/firrtl/issues/2168

There is a workaround tho.