chipsalliance/chisel #1847: Make Gitter chat room publicly accessible

itviewer, 2021-04-01:
Perhaps due to private reasons, new users must log in to access the chat room (public chat rooms do not need to log in to view the history).
The trouble is that it requires permission to access the user's private github repository.
[image]
Must agree to access the private github  repository
[image]

itviewer, 2021-04-01:
It seems that only invited users can join. Isn't this a channel for discussing publicly?

itviewer, 2021-04-01:
My goodness! see https://github.com/chipsalliance/chisel3/pull/1848

closed 2021-04-01 by itviewer

seldridge, 2021-04-01:
Ack. I don't think I realized that gitter wasn't handling the name change. Sorry about the confusion.