chipsalliance/chisel #1871: ChiselEnum: cannot use .asUInt.litValue to get the litValue

edwardcwang, 2021-04-21:
**Type of issue**: bug report / feature request

**Impact**: API addition (no impact on existing code)

**Development Phase**: request

**Other information**:

When using `ChiselEnum`, one should be able to take enum literals and use `asUInt` followed by `litValue`. This pattern is notably useful when trying to turn enum literals into `BitPat`s.

```scala
object ExampleEnum extends ChiselEnum {
  val Foo, Bar = Value
}

// Does not work since BitPat() internally calls .litValue which crashes with None.get
val simple = BitPat(ExampleEnum.Foo.asUInt)

// Instead we have to use a clumsy workaround
val complicated = BitPat(ExampleEnum.Foo.litValue.U(ExampleEnum.getWidth.W))
```

ekiwi, 2021-04-21:
This sounds like a good idea. Would you be interested in making a PR?

carlosedp, 2022-01-10:
Related to https://github.com/chipsalliance/chisel3/pull/2327