chipsalliance/chisel #1889: Empty error message for uninitialized IO. Very confusing.

armleo, 2021-04-28:
**Type of issue**: bug report
**Impact**: Bug needs fixing because is very confusing. It took me about 10 days to debug.

```
[info] compiling 1 Scala source to /home/armleo/Desktop/armleocpu/target/scala-2.12/classes ...
Elaborating design...
Done elaborating.
[error] 	at 
[info] CacheBackstorageTester:
[info] Cache Backstorage
[info] - should work very good (with firrtl) *** FAILED ***
[info]   java.lang.Exception: Problem with compilation
[info]   at chisel3.iotesters.setupFirrtlTerpBackend$.apply(FirrtlTerpBackend.scala:148)
[info]   at chisel3.iotesters.Driver$.$anonfun$execute$2(Driver.scala:52)
[info]   at scala.runtime.java8.JFunction0$mcZ$sp.apply(JFunction0$mcZ$sp.java:23)
[info]   at logger.Logger$.$anonfun$makeScope$2(Logger.scala:166)
[info]   at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[info]   at logger.Logger$.makeScope(Logger.scala:164)
[info]   at logger.Logger$.makeScope(Logger.scala:127)
[info]   at chisel3.iotesters.Driver$.$anonfun$execute$1(Driver.scala:39)
[info]   at scala.runtime.java8.JFunction0$mcZ$sp.apply(JFunction0$mcZ$sp.java:23)
[info]   at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[info]   ...
[info] ScalaTest
```
Expected behavior (backend = Verilator): 
```
Elaborating design...
Done elaborating.
[info] CacheBackstorageTester:
[info] Cache Backstorage
[info] - should work very good (with firrtl) *** FAILED ***
[info]   firrtl.passes.CheckInitialization$RefNotInitializedException: : [module CacheBackstorage]  Reference io is not fully initialized.
[info]    : io.s1.valid_out <= VOID
[info]   ...
[info] ScalaTest

```

Steps to reproduce: IO without value assignment and FIRRTL as backend

**What is the current behavior?**
See stack trace above

**What is the expected behavior?**
See stack trace number 2

Latest chisel3 release 3.4.3
Manjaro with latest updates. SBT latest form Manjaro repository.

**What is the use case for changing the behavior?**
I don't know what I am supposed to answer

jackkoenig, 2021-04-29:
@chick it looks like the `firrtl_interpreter` backend is swallowing errors from FIRRTL

whutddk, 2021-05-07:
Any suggestion? I am meeting the same bug in 3.4.2, too. the error is empty and I don't know how to locate it.

armleo, 2021-05-07:
Change backend to Verilator or something that is not firrtl.

chick, 2021-05-19:
@armleo do you have a small sample chisel example that creates this error, thanks

armleo, 2021-05-19:
@chick https://github.com/armleo/firrtl_backend_confusing_error_reproduction

chick, 2021-05-19:
@armleo That's great.
Thanks for the example. chisel-testers pr #316 will, when merged, return this message (just like verilator backend)
```Elaborating design...
Done elaborating.

 : [module firrtl_backend_confusing_error_reproduction_module]  Reference io is not fully initialized.
   : io.some_out <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module firrtl_backend_confusing_error_reproduction_module]  Reference io is not fully initialized.
   : io.some_out <= VOID
```

armleo, 2021-05-20:
Cool! Thanks a lot!

closed 2021-05-20 by ekiwi