chipsalliance/chisel #1890: Enabling Chisel plugin side-effects & questions

johnsbrew, 2021-04-29:
**Type of issue**: bug report & other enhancement
**Impact**:  unknown
**Development Phase**: request

**Other information**

When upgrading to chisel 3.4.3, we now encounter this message (line breaks added manually) about chisel plugin, instead of only its last part we were used to (often easy to fix):
```
[error] (run-main-0) chisel3.AutoClonetypeException: Unable to automatically infer cloneType on class MyClass. 
cloneType is now implemented by the Chisel compiler plugin so please ensure you are using it in your build. 
If you cannot use the compiler plugin or you are using it and you still see this message, please file an issue and let us know. 
For those not using the plugin, here is the 'runtime reflection' cloneType error message: constructor parameters (data) have values that are hardware types, which is likely to cause subtle errors. Use chisel types instead: use the value before it is turned to a hardware type (with Wire(...), Reg(...), etc) or use chiselTypeOf(...) to extract the chisel type.
```
This is perfectly fine & understandable so I decided to upgrade our repos (maybe missing a small pointer to the build.sbt example I finally found in the chisel-template repo --I was missing the scala option)
For anyone reading this for this reason before 3.5:
https://github.com/freechipsproject/chisel-template/blob/main/build.sbt#L21-L23
 

### Enabling chisel plugin side-effect

But when compiling, our code crashed elsewhere with a not so cryptic error, but **without file:line annotation**, so I had basically no clue about the actual issue (even running `last Compile / compileIncremental` for further informations)

```
[error] (Compile / compileIncremental) scala.reflect.internal.Types$TypeError: 
inferred type arguments [Nothing,S] do not conform to class MyBundle's type parameter bounds [D <: chisel3.Data,S <: ClassNotExtendingDataButBehavingAsMetaHardware[D]]
```

As the culprit class types looked weird anyway, I fixed it that way:
before:
```scala
class MyBundle[D <: Data, S <: ClassNotExtendingDataButBehavingAsMetaHardware[D]](val fakeBundle: S) extends Bundle {
  val sigA = fakeBundle.sigA
  val sigB = fakeBundle.sigB

  override def cloneType = new MyBundle[D, S](fakeBundle).asInstanceOf[this.type]
}
```
after
before:
```scala
class MyBundle[D <: Data](val fakeBundle: ClassNotExtendingDataButBehavingAsMetaHardware[D]) extends Bundle {
  val sigA = fakeBundle.sigA
  val sigB = fakeBundle.sigB

  override def cloneType = new MyBundle[D](fakeBundle).asInstanceOf[this.type]
}
// and fixing all MyBundle[D,S] into MyBundle[D] elsewhere in the code
```
None of these 2 implementations caused any issues at elaboration time in our test-cases (note: this section of our codebase is quite experimental, not used in production and hence poorly tested) 

note: `ClassNotExtendingDataButBehavingAsMetaHardware` defines a `cloneType` method but renaming it didn't change anything

### Back to our original AutoClonetypeException
The original culprit was:
```scala
class ExtendedData[D <: Data](val data: D) extends Bundle {
  val field = Bool()
}
```
originally fixed that way:
```scala
class ExtendedData[D <: Data](private val d: D) extends Bundle {
  val field = Bool()
  val data = d.cloneType
}
```

But with chisel-plugin enabled, there is no more exception raised here... is this a feature or a bug?
I think it is a feature as it means less code and no more runtime AutoClonetypeException which is a quite frightening error for beginners :)
However the second form forces the `cloneType` on the argument which I like because it isolates & never trusts some user-provided `Data` as fresh, clean & unbound which I personally prefer to the mantra "never pass bound `Data`".
Indeed the later is not applicable as soon as you deal with advanced usage where bound data is sometimes encapsulated and hence passed as arguments of larger classes. 
(Note that we do NOT use `chiselTypeOf` in place of `cloneType` as recommended because it voluntary requires bound data https://github.com/chipsalliance/chisel3/blob/v3.4.2/core/src/main/scala/chisel3/Data.scala#L240, hence killing the idea of always providing safety...)

### Summary
- No file/line pointer when debugging a chisel-plugin related type error, is there a way to improve this?
- Enabling chisel plugin raised a weird side-effect, easily fixed in our case
- Is it now always safe to pass bound-hardware as parameter of a class extending `Bundle`?

This is not a big deal anyway, thanks in advance for taking the time to explain me in further details what changed :)

johnsbrew, 2021-04-29:
@jackkoenig additional question: is the plugin intended to replace `@chiselName` macro ? I understood that from your reply to a previous issue : https://github.com/chipsalliance/chisel3/issues/1403#issuecomment-737453422

Because I started to remove some `@chiselName` and comparing the generated verilog show me some missing names.

jackkoenig, 2021-04-29:
@johnsbrew thank you for this detailed bug report and for really digging in to the issues!

> sorry @jackkoenig 😬 this was detailed in the release note https://github.com/chipsalliance/chisel3/releases/tag/v3.4.3

It is indeed, but I should probably do more outreach here. I need to write up a doc on Autoclonetype 2 (combined with some info from the current [Naming Explanation](https://www.chisel-lang.org/chisel3/docs/explanations/naming.html) which talks about the plugin a bit) and point people to it on chisel-users, Gitter, and Twitter.

> No file/line pointer when debugging a chisel-plugin related type error, is there a way to improve this?

This is absolutely a bug that needs to be fixed, thank you so much for root causing it which I imagine was pretty difficult without source locators. While I agree your code is a little funky, it obviously needs to not crash 🙂. I've reproduced the bug, working 3.4.3 with autoclonetype2 disabled (https://scastie.scala-lang.org/AHe8v359S0iHca0O7yhWlQ), failing 3.4.3 with autoclonetype enabled (https://scastie.scala-lang.org/vGRFbOS2TyuUIeNpC1HQOg).

The autoclonetype generation is probably struggling with the higher-kinded type parameter (`B <: DataGen[A]`) and generating code that later causes a type error.

> note: ClassNotExtendingDataButBehavingAsMetaHardware defines a cloneType method but renaming it didn't change anything

This doesn't surprise me, `cloneType` only matters to chisel3 when defined on a subclass of `Data`, it's irrelevant for any other types.

> Is it now always safe to pass bound-hardware as parameter of a class extending Bundle?

That is very much unintentional and another bug. Autoclonetype2 has special support for `Data` field arguments (eg. your original `val data`), but they still should be required to be unbound types. Here's a Scastie confirming the bug: https://scastie.scala-lang.org/DOeXfeVUQWiRcnmeB0JstQ

> is the plugin intended to replace @chiselName macro ?

It is! This has been true since 3.4.0 (although the behavior was refined quite a bit in 3.4.1), see the docs page https://www.chisel-lang.org/chisel3/docs/explanations/naming.html. 

> Because I started to remove some @chiselName and comparing the generated verilog show me some missing names.

This sounds like a bug, is it possible to provide a small example of what's going on? Or if the code is open-source a pointer would also work (although small examples like your others in this issue are best 🙂 )

Thanks again @johnsbrew, this is the best kind of user feedback a developer can wish for.

johnsbrew, 2021-04-30:
Thank you for your detailed answer @jackkoenig!

> This sounds like a bug, is it possible to provide a small example of what's going on? Or if the code is open-source a pointer would also work (although small examples like your others in this issue [...]

Sure, simple example with Counter

With chisel-plugin default, counter name is lost
https://scastie.scala-lang.org/hscWJ3CpTg2HEchRW56uXw
```scala
reg [2:0] value; // @[Counter.scala 60:40]
reg [2:0] value_1; // @[Counter.scala 60:40]
```

While with chiselName val names are preserved
https://scastie.scala-lang.org/QXrJn8hbTYqNefwKFvm4gw
```scala
reg [2:0] counter_value; // @[Counter.scala 60:40]
reg [2:0] another_value; // @[Counter.scala 60:40]
```

I found another example, more anecdotal since it does not concern hardware declarations, with a seq but I couldn't reproduce it simply on a very small example, so here is the excerpt from the code base
```scala
val seqOfIOs = seqOfModule.map(m => m.io.port) ++ Seq(nameA, nameB) 
```
with `@chiselName` names used are:
```verilog
<moduleName1>_io_port
<moduleName2>_io_port
<moduleName3>_io_port
nameA
nameB
```

without `@chiselName`
```verilog
<moduleName1>_io_port
<moduleName2>_io_port
<moduleName3>_io_port
seqOfIOs_4
seqOfIOs_5
```

Finally, of course, all the stuff playing with non module classes does not work with the plugin, see  #1209

mwachs5, 2021-11-02:
@jackkoenig , should we just say this is how things work without chiselName and go ahead and deprecate chiselName? Or do we have a way to restore this behavior...