chipsalliance/chisel #1901: Chisel compiled to system verilog

Z-KN, 2021-05-06:
Does Chisel compiler or FIRRTL support the conversion or the compilation to System Verilog besides Verilog?

ekiwi, 2021-05-06:
The Verilog that `firrtl` generates is also valid SystemVerilog afaik. The only difference currently is that since concurrent assertions or cover statements are not part of Verilog, we only emit these if you specify `-E sverilog`, otherwise they do get removed.
In general the aim of the `firrtl` compiler is to produce an output that works well with a large range of tools. For example all outputs will work with the open source `yosys` tool. Thus we stick to a core feature set that is enough to express every possible Chisel circuit but does not include anything fancy that tools might struggle with.

cg31, 2021-05-20:
Chisel can emit SystemVerilog:
https://github.com/chipsalliance/chisel3/blob/master/src/main/scala/chisel3/stage/ChiselStage.scala#L116

colin4124, 2021-07-06:
@cg31 

The two methods is the same,  except its name.

https://github.com/chipsalliance/chisel3/blob/4b7499f7c6287c696111bd7c6ee060f33f667419/src/main/scala/chisel3/stage/ChiselStage.scala#L96-L108

https://github.com/chipsalliance/chisel3/blob/4b7499f7c6287c696111bd7c6ee060f33f667419/src/main/scala/chisel3/stage/ChiselStage.scala#L116-L129