chipsalliance/chisel #1932: Documentation/tutorial on integrating design with bus.

m-kru, 2021-05-23:
I struggle to find a documentation or tutorial on how to integrated the design with bus (such as AXI or Wishbone). I guess there must be one as Chisel claims to allow "SoC implementations.

By the way, is it possible to automatically generate control/status registers with Chisel?

sequencer, 2021-05-23:
Maybe you wanna checkout diplomacy with RocketChip.
See https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/diplomacy/LazyModule.scala

m-kru, 2021-05-23:
@sequencer Looks like it is not what I am looking for.  Maybe I will formulate a bit different question. How does one interact with the design from the software? The usual approach is that there is some kind of bus and register map.