chipsalliance/chisel #1939: Verilator does not output "EXPECT AT" type of logs like firrtl

armleo, 2021-05-26:
**Type of issue**: bug report

**Impact**: unknown
Steps to reproduce:
Run anything that causes iotesters to fail expect

**What is the current behavior?**
No message on which cycle error happened and what was expected and what value got.
Verilator:
```
**skipped**
[info] [2.011] ReadShared (len=8) start: addr=0x068, lock=0
[info] [2.027] 8
[info] [2.139] Return dataReadShared (len=8) start: addr=0x068, lock=0
Enabling waves..
Exit Code: 0
[info] [2.211] Read address
[info] [2.217] RAN 87 CYCLES FAILED FIRST AT CYCLE 86
[info] CCXInterconnectTester:
[info] CCXInterconnect
[info] - should work very good (with verilator) *** FAILED ***
[info]   false was not true (ccx_interconnect_tester.scala:1376)
[info] ScalaTest
[info] Run completed in 51 seconds, 471 milliseconds.
[info] Total number of tests run: 1
[info] Suites: completed 1, aborted 0
[info] Tests: succeeded 0, failed 1, canceled 0, ignored 0, pending 0
[info] *** 1 TEST FAILED ***
[error] Failed: Total 1, Failed 1, Errors 0, Passed 0
[error] Failed tests:
[error] 	armleocpu.CCXInterconnectTester
[error] (Test / testOnly) sbt.TestsFailedException: Tests unsuccessful
[error] Total time: 79 s (01:19), completed May 26, 2021 2:26:53 PM
```

**What is the expected behavior?**
Proper message like for FIRRTL:
```
**skipped*
[info] [2.011] ReadShared (len=8) start: addr=0x068, lock=0
[info] [2.027] 8
[info] [2.139] Return dataReadShared (len=8) start: addr=0x068, lock=0
Enabling waves..
Exit Code: 0
[info] [2.211] Read address
[info] [2.217] RAN 87 CYCLES FAILED FIRST AT CYCLE 86
[info] CCXInterconnectTester:
[info] CCXInterconnect
[info] - should work very good (with verilator) *** FAILED ***
[info]   false was not true (ccx_interconnect_tester.scala:1376)
[info] ScalaTest
[info] Run completed in 51 seconds, 471 milliseconds.
[info] Total number of tests run: 1
[info] Suites: completed 1, aborted 0
[info] Tests: succeeded 0, failed 1, canceled 0, ignored 0, pending 0
[info] *** 1 TEST FAILED ***
[error] Failed: Total 1, Failed 1, Errors 0, Passed 0
[error] Failed tests:
[error] 	armleocpu.CCXInterconnectTester
[error] (Test / testOnly) sbt.TestsFailedException: Tests unsuccessful
[error] Total time: 79 s (01:19), completed May 26, 2021 2:26:53 PM

```

**What is the use case for changing the behavior?**
Being able to use verilator while writing tests

ekiwi, 2021-05-26:
Could you show us some snippets of your testing setup? Like how do you switch between Verilator and "firrtl"? What library versions are you using?

armleo, 2021-05-26:
**Could you show us some snippets of your testing setup? Like how do you switch between Verilator and "firrtl"?**
See repository: https://github.com/armleo/expect_at_not_showing
**What library versions are you using?**
chisel3 3.4.2
iotesters 1.5.3
Should be relevant for recent version