chipsalliance/chisel #1953: Can't have a Vec(new Bundle{}) ?

mwachs5, 2021-06-07:
**Type of issue**: bug report 

**Impact**: unknown

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
See this Scastie:
https://scastie.scala-lang.org/2qc4i466TRS9gvWIAnLmRQ

Note this is using Chisel 3.4.3 with the naming plugin enabled

**What is the current behavior?**

If I try to build a Vec(x, Bundle{}) I get a `None.get` error.

**What is the expected behavior?**

Either I get a good error message about something I am doing wrong (unclear whether Vec-of-empty-bundle is allowed, when a Reg-of-empty-bundle works fine), or it works fine and I get registers-of-empty-bundles.

**Please tell us about your environment:**

Chisel 3.4.3, with naming plugin enabled
**What is the use case for changing the behavior?**

It is nice to be able to build up Bundles of optional fields or sequential fields which might end up that the Bundle itself has nothing in it and have things just work without special casing

mwachs5, 2021-06-07:
This error is stemming from:

https://github.com/chipsalliance/chisel3/blob/master/core/src/main/scala/chisel3/Aggregate.scala#L235

```
    // Since all children are the same, we can just use the sample_element rather than all children    
    // .get is safe because None means mixed directions, we only pass 1 so that's not possible 
 direction = ActualDirection.fromChildren(Set(sample_element.direction), resolvedDirection).get
```

mwachs5, 2021-06-07:
rewriting the code to make the direction clear works to have an empty bundle in a Vec:
https://scastie.scala-lang.org/OlWvps2cRRa19sFOjdzUsg

schoeberl, 2021-09-04:
What is the meaning of a ```Reg``` or ```Vec``` of an empty ```Bundle```. Maybe ```Reg((new Bundle{})``` should err.

jackkoenig, 2021-09-18:
Example showing this in v3.4.4: https://scastie.scala-lang.org/Up44My1rRiOkIlyWjpoMLw

closed 2022-05-24 by jackkoenig