chipsalliance/chisel #2010: Cat rename signal name if it doesn't suggest name yet

colin4124, 2021-07-06:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: request

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

```scala
package chisel3
  
import chisel3._
import chisel3.stage.ChiselStage
import chisel3.util._

class Top extends RawModule {
  val in0 = IO(Input(UInt(1.W)))
  val in1 = IO(Input(UInt(1.W)))
  val in2 = IO(Input(UInt(1.W)))
  val out = IO(Output(UInt(3.W)))

  out := Cat(in2, in1, in0)
}

object Main extends App {
  (new ChiselStage).emitVerilog(new Top, args)
}
```

**What is the current behavior?**

```verilog
module Top(
  input        lo,
  input        hi_lo,
  input        hi_hi,
  output [2:0] out
);
  wire [1:0] hi = {hi_hi,hi_lo}; // @[Cat.scala 30:58]
  assign out = {hi,lo}; // @[Cat.scala 30:58]
endmodule
```

**What is the expected behavior?**

I found that `SeqUtils` would suggest named signal if it hasn't `suggested_name` yet.

https://github.com/chipsalliance/chisel3/blob/4b7499f7c6287c696111bd7c6ee060f33f667419/core/src/main/scala/chisel3/SeqUtils.scala#L29-L35

I wonder why would rename signal in `SeqUtils`

I change to as follow:

```scala
      val lo = asUInt(in.slice(0, in.length/2))
      val hi = asUInt(in.slice(in.length/2, in.length))
      hi ## lo
```

It generates:

```verilog
module Top(
  input        in0,
  input        in1,
  input        in2,
  output [2:0] out
);
  wire [1:0] _T = {in2,in1}; // @[Cat.scala 30:58]
  assign out = {_T,in0}; // @[Cat.scala 30:58]
endmodule
```
**Please tell us about your environment:**

- version: `3.5-SNAPSHOT`
- OS: `Ubuntu 18.04`

**What is the use case for changing the behavior?**

colin4124, 2021-07-06:
Here is the [demo]( https://scastie.scala-lang.org/VbQKYqkbSTKptqEv0OnlrQ)

colin4124, 2021-07-06:
Take an other critical error as example:

```scala
class Top extends RawModule {
  val in0 = IO(Input(UInt(1.W)))
  val in1 = IO(Input(UInt(1.W)))
  val in2 = IO(Input(UInt(1.W)))
  val in3 = IO(Input(UInt(1.W)))
  val out1 = IO(Output(UInt(2.W)))
  val out2 = IO(Output(UInt(2.W)))

  out1 := Cat(in1, in0)
  out2 := Cat(in3, in2)
}
```
`Cat(in3, in2)` rename `in3` to `hi`,  `in2` to `low`,  `Cat(in1, in0)` rename `in1` to `hi`,  `in0` to `low`, thus:

```shell
Elaborating design...
[error] DeletedWrapper.scala:38: Unable to name port UInt<1>(IO lo_1 in Top) to "lo" in Playground$Top@5c96796a, name is already taken by another port! in class firrtl.options.phases.DeletedWrapper
[error] DeletedWrapper.scala:38: Unable to name port UInt<1>(IO hi_1 in Top) to "hi" in Playground$Top@5c96796a, name is already taken by another port! in class firrtl.options.phases.DeletedWrapper
[error] There were 2 error(s) during hardware elaboration.
```

Here is the [demo](https://scastie.scala-lang.org/2qjBgEcARhuM4f5IlXICmw)

kammoh, 2021-07-06:
I'm curious, why is`Cat` implemented [this way](https://github.com/chipsalliance/chisel3/blob/4b7499f7c6287c696111bd7c6ee060f33f667419/src/main/scala/chisel3/util/Cat.scala) using `SeqUtils.asUInt`, rather than a simple reduce on `##`?

jackkoenig, 2021-07-06:
This is mea culpa. In the refinements to the naming plugin behavior in [v3.4.1](https://github.com/chipsalliance/chisel3/releases/tag/v3.4.1), I changed who `autoNameRecursively` works which required basically raising the priority of `.autoSeed` for when it's used in user code. The mistake was that we also use `.autoSeed` ourselves in `Cat`. Not 100% sure on the fix off the top of my head but that's the source of this bug.

jackkoenig, 2021-07-06:
> I'm curious, why is`Cat` implemented [this way](https://github.com/chipsalliance/chisel3/blob/4b7499f7c6287c696111bd7c6ee060f33f667419/src/main/scala/chisel3/util/Cat.scala) using `SeqUtils.asUInt`, rather than a simple reduce on `##`?

`.reduce` is a linear reduce and this is doing a tree reduce. We do have `.reduceTree` defined on `Vec` now though so it might be possible to use that though.

colin4124, 2021-07-07:
@jackkoenig I think your original intention is to name intermediate node to `hi` or `lo` prefix when use `SeqUtils.asUInt ` for a sequence. So it should check the `in.slice()` whether it is a intermediate node.

For example, the old chisel version:

```scala
class Top extends RawModule {
  val in0 = IO(Input(UInt(1.W)))
  val in1 = IO(Input(UInt(1.W)))
  val in2 = IO(Input(UInt(1.W)))
  val in3 = IO(Input(UInt(1.W)))
  val out = IO(Output(UInt(4.W)))

  out := Cat(in3, in2, in1, in0)
}
```

Old version generates:

```verilog
module Top(
  input        in0,
  input        in1,
  input        in2,
  input        in3,
  output [3:0] out
);
  wire [1:0] _T; // @[Cat.scala 29:58]
  wire [1:0] _T_1; // @[Cat.scala 29:58]
  assign _T = {in1,in0}; // @[Cat.scala 29:58]
  assign _T_1 = {in3,in2}; // @[Cat.scala 29:58]
  assign out = {_T_1,_T}; // @[Top.scala 15:7]
endmodule
```

New version should be:

```verilog
module Top(
  input        in0,
  input        in1,
  input        in2,
  input        in3,
  output [3:0] out
);
  wire [1:0] lo = {in1,in0}; // @[Cat.scala 30:58]
  wire [1:0] hi = {in3,in2}; // @[Cat.scala 30:58]
  assign out = {hi,lo}; // @[Cat.scala 30:58]
endmodule
```

kammoh, 2021-07-07:
> 
> `.reduce` is a linear reduce and this is doing a tree reduce. We do have `.reduceTree` defined on `Vec` now though so it might be possible to use that though.

well yeah by _reduction_ I didn't mean necessarily a linear `reduce`, though I'm pretty sure it makes no difference in synthesis QoR, and also the current "lo"/"hi" re-namings IMO don't help much with the readability of the generated Verilog either.
If FIRRTL supported "n-ary" operands, we could either directly map this to an "n-ary cat" or combine it back in a low firrlt pass and then emit 

```verilog
assign out = {in0, in1, in2, in3}
```

I can imagine the complications that "n-ary" ops can bring about, but on the other hand there are potential use cases which can benefit from them, including concatination, conditional/mux/mux-case to  Verilog `case` statements, and more efficient (QoR) n-ary arithmetic operations (such as explicit mapping to [ternary adders](http://www.martin-kumm.de/wiki/doku.php?id=04FPGA_Cores:Ternary_Adder)).

kammoh, 2021-07-07:
> @jackkoenig I think your original intention is to name intermediate node to `hi` or `lo` prefix when use `SeqUtils.asUInt ` for a sequence. So it should check the `in.slice()` whether it is a intermediate node.
> 
Even with the current binary ops, it should be able to add a lower optimization pass to remove intermediates nodes and generate something like:
```verilog
assign out = {{in3,in2},{in1,in0}};
```
To me this is more readable than the current lo/hi naming, especially when the number of concatenated signals increases.

ekiwi, 2021-07-07:
> Even with the current binary ops, it should be able to add a lower optimization pass to remove intermediates nodes and generate something like:

I thought we already had a pass that does that. @jackkoenig , do you know why it doesn't seem to do anything in this case?
https://github.com/chipsalliance/firrtl/blob/master/src/main/scala/firrtl/transforms/CombineCats.scala#L53

jackkoenig, 2021-07-08:
> I thought we already had a pass that does that. @jackkoenig , do you know why it doesn't seem to do anything in this case? [...]

Not sure, it *should* be.

PR #2011 (2021-07-14): Fix Cat rename signal #2010
### Contributor Checklist

- [ ] Did you add Scaladoc to every public function/method?
- [ ] Did you add at least one test demonstrating the PR?
- [x] Did you delete any extraneous printlns/debugging code?
- [x] Did you specify the type of improvement?
- [ ] Did you add appropriate documentation in `docs/src`?
- [x] Did you state the API impact?
- [x] Did you specify the code generation impact?
- [x] Did you request a desired merge strategy?
- [x] Did you add text to be included in the Release Notes for this change?

#### Type of Improvement

- bug fix

#### API Imp [... cut]
closed 2021-07-14 by jackkoenig