chipsalliance/chisel #2061: Change the error message about "Source is unreadable from current module"

mwachs5, 2021-08-04:
**Type of issue**: feature request

**Impact**: no functional change

**Development Phase**: request 

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

Not a bug, just suboptimal error reporting

**What is the current behavior?**

If a user does something which would be the equivalent of a "cross module reference" in Verilog, they get an error like "source is unreadable from current Module". 

**What is the expected behavior?**

We could change this error message to specifically say something like "You are attempting to do something like a cross-module reference. You probably mean to create and connect to I/Os".

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

Most of the time, when explaining this to users, we use the term "cross module reference" to explain what wrong-thing they are trying to do. We could cut out some FAQ by enhancing the error message.

sequencer, 2021-08-08:
Another idea is automatically wiring these signal by BoringUtil via FIRRTL.

anayo-hub, 2021-10-10:
I would like to make a contribution here but I need help

mwachs5, 2021-10-10:
Hi @anayo-hub, someone (Aditya) is looking at this one and turns out it is actually more involved than we realized. I think a better first issue would be #2157

PR #2248 (2021-12-09): Better MonoConnect error messages
Better error messages for MonoConnect connection errors. Fixes #2061 

### Contributor Checklist

- [ ] Did you add Scaladoc to every public function/method?
- [x] Did you add at least one test demonstrating the PR?
- [x] Did you delete any extraneous printlns/debugging code?
- [x] Did you specify the type of improvement?
- [ ] Did you add appropriate documentation in `docs/src`?
- [x] Did you state the API impact?
- [x] Did you specify the code generation impact?
- [x] Did you request a desired merge strategy?
- [ ] Did you add text to be included in the Rele [... cut]
closed 2021-12-09 by jackkoenig