chipsalliance/chisel #2068: Improve documentation for subclasses of Bundle

Freax13, 2021-08-14:
**Type of issue**: documentation

**Impact**: no functional change

**Development Phase**: proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
N/A

**What is the current behavior?**
Most subclasses of `Bundle` (like those in `chisel3.utils`) have almost all their documentation in the same group.

Take the documentation of `ArbiterIO` as an example:
[image]
The most relevant items here are `chosen`, `in`, `n` and `out` yet they are hidden among ~30 other items. This makes it very hard to find the thinks we're looking for.

**What is the expected behavior?**
The most relevant entries should be easily visible in their own group. This could be achieved either by separating the subclasses' items in their own group or by seperating the items from `Bundle` into their own group.

**Please tell us about your environment:**

N/A

**What is the use case for changing the behavior?**
Improving the documentation to improve developer experience.

jackkoenig, 2021-08-16:
This is a really good idea. ScalaDoc isn't aware of Chisel so I think the best we could do (since users may also want to run ScalaDoc on their own code without having to manually group the interesting fields for each Bundle) is to group everything inherited from `Bundle` and then the user-defined fields (and methods) are just "Ungrouped"

mwachs5, 2022-06-28:
closed by #2208

closed 2022-06-28 by mwachs5