chipsalliance/chisel #2107: Remove usages of @chiselName, which is deprecated

mwachs5, 2021-08-31:
**Type of issue**: other enhancement

**Impact**: API modification | unknown

**Development Phase**:  proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

Currently there are many usages of `@chiselName` in the codebase, but this is deprecated and you get warnings when compiling.

**What is the expected behavior?**

We get no warnings related to using deprecated `@chiselName`.

**Please tell us about your environment:**
    - version: `master`

**What is the use case for changing the behavior?**

Fewer warnings issued when compiling. Things that are deprecated should be removed.

mwachs5, 2021-10-14:
Actually `@chiselName` is not technically deprecated, the warning that it gets is:

```
[warn] /home/runner/work/chisel3/chisel3/src/main/scala/chisel3/util/Arbiter.scala:103:2: Auto-application to `()` is deprecated. Supply the empty argument list `()` explicitly to invoke method length,
[warn] or remove the empty argument list from its definition (Java-defined methods are exempt).
[warn] In Scala 3, an unapplied method like this will be eta-expanded into a function.
[warn] @chiselName
```

Burnleydev1, 2021-10-14:
Alright @mwachs5, I'll get on it now.

Burnleydev1, 2021-10-14:
Please I need help with setting up an environment

mwachs5, 2021-10-14:
> But if I'm getting you right then what you mean is that I should not delete the @chiselName in the test repository

there is a directory src/test/scala, some of the tests are specifically for @chiselName and there we shouldn't delete @chiselName. If there any others in that directory you could ask about them in your PR

mwachs5, 2021-10-14:
Yes, please delete all the ones in `src/main/scala` like the example you gave... also in core/src/main/scala

Burnleydev1, 2021-10-14:
Okay thank you for the clarification @mwachs5.

Burnleydev1, 2021-10-14:
Hello @mwachs5, this is the link to the first pull request: https://github.com/chipsalliance/chisel3/pull/2186

mwachs5, 2023-02-08:
This was done in #2635

closed 2023-02-08 by mwachs5