chipsalliance/chisel #2128: Fail to use `addSource` for objects having the same value

retrhelo, 2021-09-17:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

A bug with `chisel3.util.experimental.BoringUtils`. Things get wrong when writing codes like this:
```Scala
class Top extends RawModule {
	val io = IO(new Bundle {
		val in = Input(Bool())
		val out1 = Output(Bool())
		val out2 = Output(Bool())
	})

	val inn1 = Module(new Inner1)
	val inn2 = Module(new Inner2)

	val value1 = WireInit(false.B)
	val value2 = WireInit(false.B)
	addSink(value1, "inner1")
	addSink(value2, "inner2")

	io.out1 := value1
	io.out2 := value2
}

class Inner1 extends RawModule {
	val io = IO(new Bundle {
		val out = Output(Bool())
	})

	val value = WireInit(true.B)

	io.out := value
	addSource(value, "inner1")
}

class Inner2 extends RawModule {
	val io = IO(new Bundle {
		val out = Output(Bool())
	})

	val value = WireInit(false.B)

	io.out := value
	addSource(value, "inner2")
}
```

Trying to generate verilog for `Top` would result in an wrong `Exception in thread "main" firrtl.passes.wiring.WiringException: Unable to determine source mapping for sink 'List(Top)'`. However, this error can be avoided by changing `value`'s value in either `Inner1` or `Inner2` like this: `val value = WireInit(true.B)`. It seems that using `addSource` for more-that-one `Wire` object having the same constant value will result in this bug. When the values are different, this error disappears. 

**What is the current behavior?**

The current behavior is that I can't use `addSource` for more-that-one object **having the same constant value**. When doing this, an error would be shown if trying to **generate verilog code**. 

**What is the expected behavior?**

I think it would be fair enough to `addSource` for different object, even if they have the same value. The code I put above should work without any problem. 

**Please tell us about your environment:**

version: 3.4.3
OS: `Linux fedora 5.13.12-200.fc34.x86_64 #1 SMP Wed Aug 18 13:27:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

**What is the use case for changing the behavior?**

It's fair enough to use `addSource` for different objects, and even the same object using different keys.

seldridge, 2021-09-17:
The issue is because if those values are the same, `Inner1` and `Inner2` will deduplicate to the same module.  You can work around this by disabling deduplication for those modules.  Try changing the `addSource` methods to:

```scala
addSource(value, "inner1", disableDedup=true)
/* ... */
addSource(value, "inner2", disableDedup=true)
```

Scastie showing this working: https://scastie.scala-lang.org/0egkINozRsyXKvsXfxDvqg

The fundamental problem is that the wiring transform needs to have a unique source driver for each sink.  If it doesn't have this, then it gets confused and errors out (with that message you see).  The `bore` API is a bit safer than `addSource`/`addSink` and adds those deduplication flags to work around this exact issue.

retrhelo, 2021-09-18:
The fundamental problem is that the wiring transform needs to have a
    unique source driver for each sink. If it doesn’t have this, then it
    gets confused and errors out (with that message you see).

Yes, I guess for a similiar reason. But I wonder why making 
|disableDuplicated = false| as the default option, I’d say that I can’t 
find a clear benefit from it…

    The bore API is a bit safer than addSource/addSink and adds those
    deduplication flags to work around this exact issue.

This is a choice, but in my opinion |bore| lacks flexibility, as it is 
called in the parent module. It will be rather difficult to make wiring 
clear once the project grows larger.

​

closed 2021-09-18 by retrhelo