chipsalliance/chisel #2143: Deprecate toBools in Bits (should be asBools)

mwachs5, 2021-10-04:
See discussion in https://github.com/chipsalliance/chisel3/pull/2124#discussion_r721610315

**Type of issue**:feature request | documentation | other enhancement

**Impact**:  API modification

**Development Phase**:  proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

There still exists a `.toBools` function, undeprecated, in Bits.scala

**What is the expected behavior?**

All the `.toBools` were deprecated and turned into `asBools`. This one seems to have been forgotten? Or was that intentional?

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

consistent function names leading to less cognitive load on users

sequencer, 2021-10-11:
I just notice this:
```
314:  final def toBools: Seq[Bool] = macro SourceInfoTransform.noArg
317:  final def toBools(dummy: Int*): Seq[Bool] = macro SourceInfoWhiteboxTransform.noArgDummy
```
https://github.com/chipsalliance/chisel3/commit/121635ed26c8a9852c827d6c0729515337604d08
didn't deprecate `toBools`, but deprecated `do_toBools`,
so I just removed `do_toBools`, but didn't remove `toBools`.

If user from 3.4 used `toBools`, they will receive a runtime deprecation warning, rather than a normal deprecation warning, (and this may break binary compatibility? @jackkoenig)

So currently, in the master code, `toBools` will introduce which will throw `do_toBools` not implemented, so I propose remove `toBools` directly in Chisel 3.5, since runtime deprecation warning has already warned users.

debs-sifive, 2021-10-19:
@ibukenya nobody is working on this ticket yet, if you want to give it a try!

jackkoenig, 2024-02-28:
This was removed in v3.5.0, fixed by https://github.com/chipsalliance/chisel/pull/2170

closed 2024-02-28 by jackkoenig