chipsalliance/chisel #2144: Vec.reduceTree produces suboptimal reduction tree

nicmcd, 2021-10-04:
**Type of issue**: bug report

**Impact**: no functional change | increase efficiency in produced verilog

**What is the current behavior?**
Vec[Bool].reduceTree produces suboptimal reductions.
```
// chisel
class OrReduce(n: Int) extends Module {
  val io = IO(new Bundle {
    val in = Input(Vec(n, Bool()))
    val out = Output(Bool())
  })
  io.out := io.in.reduceTree(_ | _)
}

// generated verilog
module OrReduce(
  input   clock,
  input   reset,
  input   io_in_0,
  input   io_in_1,
  input   io_in_2,
  input   io_in_3,
  input   io_in_4,
  input   io_in_5,
  input   io_in_6,
  input   io_in_7,
  output  io_out
);
  assign io_out = io_in_0 | io_in_1 | (io_in_2 | io_in_3) | (io_in_4 | io_in_5 | (io_in_6 | io_in_7)); // @[OrReduce.scala 13:32]                                                                                  
endmodule
```
This OR reduction has 6 gates of delay.

**What is the expected behavior?**
The produced reduction should be as follows:
```
assign io_out = ((io_in_0 | io_in_1) | (io_in_2 | io_in_3)) | ((io_in_4 | io_in_5) | (io_in_6 | io_in_7));
```
This has only 3 gate delays.

**Please tell us about your environment:**
chisel = 3.4.3
scala = 2.12.13

**What is the use case for changing the behavior?**
Bitwise reductions are very common. If a function name implies the algorithm used for reduction (i.e., reduceTree) then the produced hardware should match the named algorithm.

sequencer, 2021-11-19:
I think synthesis tool should handle this directly, since the fanout selection is actually a physical sense synthesis job, chisel doesn’t aware of physical information, which make chisel impossible to know how to generate a better circuit with optimized fanout.

nicmcd, 2021-11-19:
I know that a synthesis tool *may* optimize the reduction tree, but if Chisel is going to rely on this then logically there should be no reduceTree() function at all since the reduce() will be equivalently reduced. So the question is... what is the purpose of the reduceTree() function?

seldridge, 2021-11-19:
Several things:

1) We should fix `reduceTree()` so that it produces a better output here.   This just looks broken.  There is an alternative implementation that I'm aware of in a library that we could import: https://github.com/grebe/ofdm/blob/afd9cd404d6748185237d28edfca821b897642a3/src/main/scala/ofdm/TreeReduce.scala#L3 I haven't tested this, but it may do the right thing.

2) Chisel and the Scala FIRRTL Compiler are weird. This is relying on Chisel user coding style of a mux tree and expecting a FIRRTL compiler to respect that behavior. There should be a better mechanism to do this other than "mux trees are special" like allowing for inline Verilog if somebody _really_ cares about an exact coding style in the output or using APIs which place optimization barriers/"don't touch" around a specially coded mux tree. As a counterpoint, compiling either the `reduce` or `reduceTree` output with CIRCT (an alternative FIRRTL compiler) produces the same output you'd get from `reduce` because it is aggresively canonicalizing away intermediary nodes/wires.

3) Annoyingly, backend tools may really care about the mux tree structure. Yosys is known to not be good at optimizing these and will report higher gate counts/frequencies if you use one over the other. This is a reason why we have `reduceTree`. However, this should really be an option to your FIRRTL compiler. E.g., `firrtl --tool=yosys Foo.fir` should reorganize mux trees to be optimal for Yosys whereas `firrtl --tool=genus Foo.fir` may not.

There was a long discussion about mux trees [here](https://github.com/chipsalliance/chisel3/issues/1199) which touches on a lot of this.

tl;dr: we should fix `reduceTree()`. We should explore vendor/tool specific optimizations in new compilers.

nicmcd, 2021-11-19:
I think it would be better if reduceTree did not exist. Is there a case where not having a tree is better than having a tree? Maybe reduce() should always produce a tree.

jackkoenig, 2021-11-19:
> I think it would be better if reduceTree did not exist. 

Possibly, sometimes it is useful for the user to have more control over exactly how the Verilog looks, although sometimes it is an antipattern.

> Is there a case where not having a tree is better than having a tree? Maybe reduce() should always produce a tree.

Yes, we have measured in the past that leaving things in a simpler, flat form sometimes results in better synthesis QoR (when compared to emitting a tree structure). I'd suggest reading @seldridge's link to the discussion about mux trees in full, but in particular I'd suggest reading [this comment](https://github.com/chipsalliance/chisel3/issues/1199#issuecomment-609509327). 

Put short, many synthesis tools do a very good job and trying to do too much in the Verilog can actually hurt the QoR.

schoeberl, 2021-11-23:
I would like to keep reduceTree. It is not only for MUXes or simple gates. You can also have a tree of more complex circuits (e.g., an arbitration tree including a register at each node). This is probably an (almost) impossible thing a synthesis tool can infer from a plain reduce.

nicmcd, 2021-11-23:
Ok, then it seems reduceTree is still desired. If reduceTree is desired, then it should absolutely make a proper tree. The effect of synthesizers optimizing the ill-formed tree is irrelevant for this case. User's can either use reduce relying fully on the synthesizer, or use reduceTree which should produce a proper tree. All agree?

seldridge, 2021-11-29:
Agree. Fix `reduceTree` so it does the right thing.

schoeberl, 2021-12-15:
I spent some time on playing around to build a binary tree for non-power of 2 numbers of elements. There is no obvious "perfect" solution (as far as I could find out), but I have a reasonable idea. I will give it a shot. Maybe that fixes also the type issue that I mentioned in: https://scastie.scala-lang.org/schoeberl/XaIIU9exRECOoMJ1vwaSSg/16
BTW, should reduceTree be an enhancing method for ```Seq''' instead of just ```Vec'''? A collection of Chisel elements does not need to be in a Vec. I actually think it should be better in a Scala collection type. Vec should be a representation of a Mux (and/or an enable demux for a collection of registers) and not much more.

jackkoenig, 2022-07-05:
Fixed by https://github.com/chipsalliance/chisel3/pull/2318

closed 2022-07-05 by jackkoenig