chipsalliance/chisel #2163: Some of the Cookbooks are missing from the list of Cookbooks

mwachs5, 2021-10-10:
**Type of issue**: documentation 

**Impact**: no functional change 

**Development Phase**: request 

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

There has been new content added (new Cookbooks), but not all are listed on the cookbooks "table of contents": https://github.com/chipsalliance/chisel3/blob/master/docs/src/cookbooks/cookbooks.md

This means they do not currently show up on the website (which was probably intentional @azidar , @jackkoenig ?) but they will also not show up once we release 3.5.0 and switch the website to refer to that.

We should look at all the cookbooks and update the Cookbook Table of Contents to make sure they are all captured.

**What is the current behavior?**

Only some of the cookbooks (in https://github.com/chipsalliance/chisel3/blob/master/docs/src/cookbooks/) are listed on: https://github.com/chipsalliance/chisel3/blob/master/docs/src/cookbooks/cookbooks.md

**What is the expected behavior?**

All of the cookbooks, especially DataView and the Hierarchy one, should be listed show up as of 3.5.

These should also show up on the sidebar at https://www.chisel-lang.org/chisel3/docs/cookbooks/cookbooks.html, this would need to be updated https://github.com/freechipsproject/www.chisel-lang.org/blob/ef93edd993b9e64a26d076626597b0ca10e3f696/docs/src/main/resources/microsite/data/menu.yml#L18 but I don't think we want to do that until after the 3.5 bump, so maybe that should be its own issue?

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

Find the new and exciting content for the 3.5 bump!

Burnleydev1, 2021-10-10:
Hello @mwachs5  When I click on the present cookbooks it gives a 404 page.

mwachs5, 2021-10-10:
@Burnleydev1 what link exactly is giving you the 404? I just checked them again (I think!)

Burnleydev1, 2021-10-10:
[image]
These ones

Burnleydev1, 2021-10-10:
[image]

Burnleydev1, 2021-10-10:
* [General Cookbooks](cookbook)
* [Naming Cookbook](naming)
* [Troubleshooting Guide](troubleshooting)

Burnleydev1, 2021-10-11:
Okay thanks for the clarification @mwachs5

Burnleydev1, 2021-10-12:
Thanks for the clarification I'll work on it right away

Burnleydev1, 2021-10-12:
> Specifically, in this directory the `hierarchy` and `naming` docs don't show up on that front page:
> 
> https://github.com/chipsalliance/chisel3/tree/master/docs/src/cookbooks

I can actually see the naming docs on my end but I'm not seeing the dataview docs

mwachs5, 2021-10-12:
ah yeah you are right i mean `hierarchy` and `dataview`.

Burnleydev1, 2021-10-12:
Oh, then its more complicated than I thought it was.

closed 2022-05-24 by mwachs5