chipsalliance/chisel #2190: Add a CONTRIBUTING.md

mwachs5, 2021-10-18:
**Type of issue**: documentation

**Impact**: no functional change

**Development Phase**: proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

It's a bit unclear how to contribute to chisel, or how to just get started as a developer. We have docs on how to set up your own chisel project, but that's not *quite* the same thing.

**What is the expected behavior?**

There should be a section in the README like "Interested in Contributing to the Project? See ...."

- Should point to the existing SETUP.md and perhaps mention something about the CLA that will need to be signed.
- Can also explain how to find/file issues and use the `good-first-issue` tag.

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

Smoother experience for new contributors

Burnleydev1, 2021-10-19:
Yes, please I will need pointers

Burnleydev1, 2021-10-19:
Though I understand that I should put a link that sends an interested contributor to the setup.md

mwachs5, 2021-10-19:
Ok. I often see other repositories with a CONTRIBUTING.md, so you can create one at the top level of this project (can do through GitHub UI). Then edit the README.md to point to it with a new heading. 

Inside we should include something like:

0. Link to GitHub guide on Pull Requests
1. How to compile and test chisel3 (link to the SETUP.md)
2. A mention that you'll need to sign the CLA
3. Anything else you found useful as a new contributor!

Burnleydev1, 2021-10-19:
Okay thanks @mwachs5 I will get on it.