chipsalliance/chisel #2233: Add guide on creating issues to CONTRIBUTING.md

Burnleydev1, 2021-11-03:
**Type of issue**: bug report | feature request | documentation | other enhancement
Documentation

**Impact**: no functional change | API addition (no impact on existing code) | API modification | unknown
No functional change

**Development Phase**: request | proposal
proposal
**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**
CONTRIBUTING.md has no guide on creating an issue
**What is the expected behavior?**
I think CONTRIBUTING.md should have a guide on creating a new issue.
**Please tell us about your environment:**

**What is the use case for changing the behavior?**

Burnleydev1, 2021-11-03:
Hello @mwachs5, can I work on this?
```
## Reporting Issues

### You have a problem

Please be so kind as to search for any open issue already covering your problem.

If you find one, comment on it so we can know there are more people experiencing it.

If not, you can go ahead and create an issue with as much detail as you can provide.

### You have a suggestion

Please be so kind as to search for any open issue already covering your suggestion.

If you find one, comment on it so we can know there are more people supporting it.

If not, you can go ahead and create an issue. 

### Look For an Existing Issue

Before you create a new issue, please do a search in [open issues](https://github.com/chipsalliance/chisel3/issues) to see if the issue or feature request has already been filed.

Be sure to scan through the [feature requests](https://github.com/chipsalliance/chisel3/issues?q=is%3Aopen+is%3Aissue+label%3A%22feature+request%22) feature requests.
or [bug fix](https://github.com/chipsalliance/chisel3/issues?q=is%3Aopen+is%3Aissue+label%3Abug)
and others depending on the type of issue.

If you find your issue already exists, comment on it so we can know there are more people supporting it.

If you cannot find an existing issue that describes your bug or feature, create a new issue with as much detail as you can.
```

Burnleydev1, 2021-11-03:
## Reporting Issues

### You have a problem

Please be so kind as to search for any open issue already covering your problem.

If you find one, comment on it so we can know there are more people experiencing it.

If not, you can go ahead and create an issue with as much detail as you can provide.

### You have a suggestion

Please be so kind as to search for any open issue already covering your suggestion.

If you find one, comment on it so we can know there are more people supporting it.

If not, you can go ahead and create an issue. 

### Look For an Existing Issue

Before you create a new issue, please do a search in [open issues](https://github.com/chipsalliance/chisel3/issues) to see if the issue or feature request has already been filed.

Be sure to scan through the [feature requests](https://github.com/chipsalliance/chisel3/issues?q=is%3Aopen+is%3Aissue+label%3A%22feature+request%22) feature requests.
or [bug fix](https://github.com/chipsalliance/chisel3/issues?q=is%3Aopen+is%3Aissue+label%3Abug)
and others depending on the type of issue.

If you find your issue already exists, comment on it so we can know there are more people supporting it.

If you cannot find an existing issue that describes your bug or feature, create a new issue with as much detail as you can.

mwachs5, 2021-11-03:
Sure... our README also encourages people to use stack overflow and gitter. We want to discourage people from opening GitHub issues for user error "I can't get it to work" type questions.

Burnleydev1, 2021-11-03:
I would like to know if this is relevant this is relevant.

mwachs5, 2021-11-03:
Sure, i think this is helpful, just want to make sure we have it in the right place and there is not conflicting information. I think when you try to open an issue it directs you to SO. Perhaps some of this content could go there?

https://raw.githubusercontent.com/chipsalliance/chisel3/master/.github/ISSUE_TEMPLATE.md

```
<!-- Note: for support questions, you are strongly advised to use [Stack Overflow](https://stackoverflow.com/questions/tagged/chisel). -->
<!--
This repository's issues are reserved for feature requests and bug reports.
Please select the item best describing the issue in each category and delete the other items.
-->
```

Actually, one thing you could do is get GitHub to stop complaining about our ISSUE_TEMPLATE being out-of-date...

Burnleydev1, 2021-11-03:
Okay please how can I go about it, I'm confused on how to go about the ISSUE_TEMPLATE being out-of-date
Should I update the syntax? 

> Sure, i think this is helpful, just want to make sure we have it in the right place and there is not conflicting information. I think when you try to open an issue it directs you to SO. Perhaps some of this content could go there?
> 
> https://raw.githubusercontent.com/chipsalliance/chisel3/master/.github/ISSUE_TEMPLATE.md
> 
> ```
> 
> 
> ```
> 
> Actually, one thing you could do is get GitHub to stop complaining about our ISSUE_TEMPLATE being out-of-date...

Burnleydev1, 2021-11-03:
Okay I've seen a documentation on from the [learn more](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/about-issue-and-pull-request-templates) link.

Burnleydev1, 2021-11-05:
Should I close this issue, or should I work on the contributing.md file, making sure the information from it and the new issue template being created is not conflicting?

Burnleydev1, 2021-11-06:
> Hello @mwachs5, Should I close this issue, or should I work on the contributing.md file, making sure the information from it and the new issue template being created is not conflicting?

What I mean is if we still need to add the guide to creating issues?

mwachs5, 2021-11-06:
> What I mean is if we still need to add the guide to creating issues?

What do you think? As a newcomper to the community, was it unclear how to create issues?

Burnleydev1, 2021-11-06:
> 
> What do you think? As a newcomper to the community, was it unclear how to create issues?

No it wasn’t unclear how to create.

closed 2021-11-06 by Burnleydev1