chipsalliance/chisel #2236: GitHub complains our ISSUE_TEMPLATE is out of date

mwachs5, 2021-11-03:
**Type of issue**:  other enhancement

**Impact**: no functional change

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

GitHub complains about our current ISSUE_TEMPLATE:
[image]

**What is the expected behavior?**

GitHub does not complain about our issue template

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

I'm not sure. SOmeone should click that "read more" link to see what goodness we could be getting by upgrading our ISSUE_TEMPLATE

Burnleydev1, 2021-11-03:
Hi @mwachs5, please can I work on this?

Burnleydev1, 2021-11-03:
I actually think I understand how the new IISUE_TEMPLATE new issue template is created.
should custom Issue template be part of the the options for this project?
[image]

mwachs5, 2021-11-03:
no, we should clean up the bug report / feature request ones and not offer custom issue template to peopel filing issues

Burnleydev1, 2021-11-03:
Okay @mwachs5 thank you, I think I will need to do the changes from my cloned repo and do a pull request.

Burnleydev1, 2021-11-03:
Hello @mwachs5, I'm unable to do those changes, probably because it does not work for forked repositories

Burnleydev1, 2021-11-03:
It works, my browser was glitching.

Burnleydev1, 2021-11-03:
Hello @mwachs5, I was thinking of adding documentation and other enhancements too, as it is possible to do so.

Burnleydev1, 2021-11-04:
This is pull request on this issue: https://github.com/chipsalliance/chisel3/pull/2237

mwachs5, 2022-06-28:
closed by #2237 (and fixed up by #2609 ). Thank you @Burnleydev1 !

closed 2022-06-28 by mwachs5