chipsalliance/chisel #2242: Make AOP being able to remove IR

sequencer, 2021-11-08:
**Type of issue**: feature request 

**Impact**: no functional change | API addition (no impact on existing code) | API modification | unknown

**Development Phase**: proposal

`InjectingTransform` can inject ChiselIR after elaboration, see `chiselTests.aop.InjectionSpec`, while I found there is no possibility to delete IR with AOP, this is a little annoying: to achieve this, I have to write a FIRRTL transform, however firrtl transform cannot use chisel, and loss some metadata.

seldridge, 2021-11-08:
This sounds very hard to implement. 😅 

I am slowly developing the opinion that `InjectingAspect` and `InjectingTransform` are likely considered harmful (in their existing implementations). We pack a bunch of IR into an annotation and then append it during a transform. This feels weird. There are also problems with injection not working (and not planning to be supported) for the Definition/Instance and Instantiate APIs.

I think what the real ask here is for a simpler API to write simple transforms without having to do FIRRTL traversal (recursive AST walking with a ton of boilerplate and user overhead like maintaining a rename map). Such an API could either be an actual FIRRTL transform API (like MLIR declarative rewrite rules, e.g., [replacing `mux(invalid, x) -> x` in a pretty high level way](https://github.com/llvm/circt/pull/2099/files)) or higher level inside Chisel.

Do you have some examples of things you would like to be able to do with such an API that you're proposing here @sequencer?  Maybe we could discuss that independent of an AOP-style implementation. 😄

sequencer, 2021-11-19:
After carefully thinking about this I do believe this is not a good API.

closed 2021-11-19 by sequencer