chipsalliance/chisel #2253: Refactor Queue with dataview+Definition/instance API

sequencer, 2021-11-19:
**Type of issue**: feature request

**Impact**: API modification

**Development Phase**: proposal

**What is the current behavior?**
A lot of queue will be generated in a large-scale design, which was known as the `Queue` issue in rocketchip.

**What is the expected behavior?**
Trying to use the new feature in Chisel 3.5 to resolve this. Only generate one module for each same queue. 
Like what I did in the #2250, I’d like to refactor queue with definition/instance api to cache queue with same api. 
However the problem is Queue Module consumes Chisel data type as a type parameter which makes this impossible to implement, since it’s each queue almost has different type parameters.
But actually queue didn’t use the chisel type parameter, it only use which to do the sanity check inside queue, which is not related to the real queue hardware implementation. 
Thus I’d like to propose a API breaking change to move those changes out from Queue Module, while provide a same user  interface via dataview API. 
This will break `new Queue(…)` API.
But the Queue factory will remain the same.
 
**Please tell us about your environment:**
    - version: `3.5-SNAPSHOT`
**What is the use case for changing the behavior?**
Optimize the generated verilog code to use less Queue Module, and speed up elaborate pass a little, as well as dedup pass.

mwachs5, 2021-11-19:
Note that once you start using Instances in your design, you have a lot of
downstream impact. For example any code using the Select APIs needs to be
rewritten.

sequencer, 2021-11-19:
Yes.
I think we can still select an instance and annotate via `toAbsoluteTarget`? This should be a breaking change as well.

mwachs5, 2021-11-19:
> I think we can still select an instance

You need to use a different API to select instances. The normal `.collectDeep` will fail when it encounters an `Instance`. See https://github.com/chipsalliance/chisel3/blob/master/docs/src/cookbooks/hierarchy.md#how-do-I-use-the-new-hierarchy-specific-Select-functions

sequencer, 2021-11-19:
Sounds reasonable, at least AOP still works with a different API.
This modification should be a breaking change. I think we to discuss the roadmap to implement this.
In my point of view, definition might not suite for normal user, but it’s a very good tool for library writers.

mwachs5, 2021-11-19:
> definition might not suite for normal user, but it’s a very good tool for library writers.

I disagree. Users are the ones who know that they want to make "N of a thing". The can always call `Definition` on the library thing. 

I think this is something interesting to experiment with but I don't think we can do this for `Queue` *yet*

sequencer, 2021-11-19:
Got it. Another idea is maintaining another util factory in the hierarchy package, user can freely decide which package should use(normal Queue or Instance Queue)

sequencer, 2021-11-20:
And maybe it's possible to add a compile option to ask chisel standard library to use the instance version of libraries.