chipsalliance/chisel #2255: sbt test fails

schoeberl, 2021-11-23:
**Type of issue**: bug report 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

Running from HEAD on master a plain

```
sbt compile
sbt test
```
fails as follows:

```
[error] /Users/martin/tmp/chisel3/src/test/scala/chiselTests/IntervalSpec.scala:18:8: object FIRRTLException is not a member of package firrtl
[error] import firrtl.{FIRRTLException, HighFirrtlCompiler, LowFirrtlCompiler, MiddleFirrtlCompiler, MinimumVerilogCompiler, NoneCompiler, SystemVerilogCompiler, VerilogCompiler}
[error]   
```

ekiwi, 2021-11-23:
CI also fails: https://github.com/chipsalliance/chisel3/runs/4293801968?check_suite_focus=true

@jackkoenig @sequencer

closed 2021-11-23 by mergify[bot]

schoeberl, 2021-11-24:
Thanks for the quick fix. Should have done this myself...