chipsalliance/chisel #2286: feature: scala3 support

borissmidt, 2021-12-04:
Dear,

I'm a scala developer and i was wondering if there are any plans on upgrading the scala version and go to scala-3.

I dont really do hardware development but it seems like a very cool project.

mwachs5, 2021-12-04:
Hi @borissmidt , it's certainly on our radar as something to explore. We haven't yet investigated what it would take.

sequencer, 2021-12-05:
The tricky things are macro and compiler plugin, both of which are tightly integrated to Chisel. I'm also experimentally splitting a Scala3 & Scala213 code bases in #2285, but not confident in getting this into Chisel.
I'm trying to firstly split all  `macro` out of chisel code base, and moving them into a Scala2 src folder, then implementing a Scala3 version of them to minimize the Scala2 and Scala3 gap.
Compiler seems easier than macro(only need to rewrite them in Scala).

cnmade, 2022-02-18:
+1 scala3, scala3 provided better API , library and optimze much more than scala 2.x

jnbrq, 2022-09-20:
I think Scala3 support would elevate Chisel's reach and ease of usability.

counter2015, 2025-01-06:
Is there any roadmap or updates for Scala 3 support ? Since I have found some [commit](https://github.com/chipsalliance/chisel/pull/4591) related to Scala 3 had been merged.

sequencer, 2025-01-06:
We are trying multiple different ways go to Scala 3.
I also designed the MVP of Scala 3 eDSL target to MLIR in http://github.com/sequencer/zaozi
Note this would not be backwards compatible, it just provides an example to design a circuit eDSL with Scala + CIRCT.