chipsalliance/chisel #2287: Using enums in Chisel3 - not found: value switch

Fleker, 2021-12-07:
**Type of issue**: documentation

**Impact**: no functional change

**Development Phase**: request | proposal

**Other information**
I'm following the [chisel enum guide](https://www.chisel-lang.org/chisel3/docs/explanations/chisel-enum.html#functionality-and-examples), which shows how to use a switch statement on an enum.

```scala
class AluMux1File extends Module {
  val io = IO(new AluMux1Bundle)

  // Default value for aluMux1Out
  io.aluMux1Out := 0.U

  switch (io.aluMux1Sel) {
    is (selectRS1) {
      io.aluMux1Out := io.rs1Out
    }
    is (selectPC) {
      io.aluMux1Out := io.pcOut
    }
  }
}
```

But when I try to do this in my own code, I get compiler errors:

```
error] /home/.../MyModule.scala:421:5: not found: value switch
[error]     switch (io.opcode) {
[error]     ^
[error] /home/.../MyModule.scala:422:9: not found: value is
[error]         is (noop) {
[error]         ^
[error] /home/.../MyModule.scala:422:13: not found: value noop
[error]         is (noop) {
[error]             ^
[error] /home/.../MyModule.scala:425:9: not found: value is
[error]         is (inv) {
[error]         ^
[error] /home/.../MyModule.scala:425:13: not found: value inv
[error]         is (inv) {
[error]             ^
[error] 5 errors found
```

It seems like [the syntax in Scala](https://alvinalexander.com/scala/how-to-use-scala-match-expression-like-switch-case-statement/) is a bit different, so the code snippet should be instead:

```scala
import AluMux1Sel._

class AluMux1Bundle extends Bundle {
  val aluMux1Sel = Input(AluMux1Sel())
  val rs1Out     = Input(Bits(32.W))
  val pcOut      = Input(Bits(32.W))
  val aluMux1Out = Output(Bits(32.W))
}

class AluMux1File extends Module {
  val io = IO(new AluMux1Bundle)

  // Default value for aluMux1Out
  io.aluMux1Out := 0.U

  io.aluMux1Sel match {
    case AluMux1Sel.selectRS1 => {
         io.aluMux1Out := io.rs1Out
    }
  }
  switch (io.aluMux1Sel) {
    is (selectRS1) {
  
    }
    is (selectPC) {
      io.aluMux1Out := io.pcOut
    }
  }
}
```

jackkoenig, 2021-12-07:
I think you're missing an `import`, `switch` and `is` are defined in package `chisel3.util`, here's a full working example:
```scala
import chisel3._
import chisel3.util.{switch, is}
import chisel3.stage.ChiselStage
import chisel3.experimental.ChiselEnum

object AluMux1Sel extends ChiselEnum {
  val selectRS1, selectPC = Value
}

import AluMux1Sel._

class AluMux1Bundle extends Bundle {
  val aluMux1Sel = Input(AluMux1Sel())
  val rs1Out     = Input(Bits(32.W))
  val pcOut      = Input(Bits(32.W))
  val aluMux1Out = Output(Bits(32.W))
}

class AluMux1File extends Module {
  val io = IO(new AluMux1Bundle)

  // Default value for aluMux1Out
  io.aluMux1Out := 0.U

  switch (io.aluMux1Sel) {
    is (selectRS1) {
      io.aluMux1Out := io.rs1Out
    }
    is (selectPC) {
      io.aluMux1Out := io.pcOut
    }
  }
}

object Main extends App {
  println(ChiselStage.emitVerilog(new AluMux1File))
}
```
Executable example: https://scastie.scala-lang.org/aLDCoh6ET3K4jKHQdlih5Q

jackkoenig, 2021-12-07:
You are correct about Scala match syntax, but that actually doesn't work for generating hardware in Chisel. Think about a Chisel design as a Scala program that when you run it, the execution of the code creates a hardware graph. We're not synthesizing the Chisel AST down to hardware so in Chisel, when the match runs, it'll only pick 1 path and ignore the rest, which wouldn't give you the hardware you want.

Fleker, 2021-12-07:
Ah, thanks for pointing this out. Will close.

closed 2021-12-07 by Fleker