chipsalliance/chisel #2307: Concerns on privatisation of chisel3.internal.firrtl._

johnsbrew, 2021-12-17:
**Type of issue**: API UNEXPECTED DELETION

**Impact**: API deletion 

**Development Phase**: request

**Other information**
#2274 quietly made almost all elements of `chisel3.internal.firrtl._` private

Even if it was probably a bad idea, we **DO** use this internal API to add a hook in between chisel elaboration and firrtl compilation, in particular to add manually some annotations, and to get some information about the components of the circuit.

Here is the main excerpt:
```scala
import chisel3.internal.{firrtl => cir}
case class ElaboratedDesign(raw: AnnotationSeq) {
  // Circuit declaration is in chisel3 core => outside main src folder => not registered in API DOC
  // For reference, here is the declaration:
  // case class Circuit(name: String, components: Seq[Component], annotations: Seq[ChiselAnnotation] = Seq.empty)

  lazy val circuit                            = raw.collectFirst { case ChiselCircuitAnnotation(c) => c }.get
  lazy val chiselAnnos: Seq[ChiselAnnotation] = circuit.annotations
  lazy val name                               = circuit.name
  lazy val components                         = circuit.components
  lazy val design = raw.collectFirst { case chisel3.stage.DesignAnnotation(d) => d } match {
    case Some(uut) => uut
    case None      => throw new Exception("Something went very wrong, no design were found")
  }

  def mapCircuit(f: cir.Circuit => cir.Circuit): ElaboratedDesign = {
    this.copy(raw = raw.map(a => {
      a match {
        case ChiselCircuitAnnotation(c) => ChiselCircuitAnnotation(f(c))
        case _                          => a
      }
    }))
  }

  def mapName(f: String => String): ElaboratedDesign = mapCircuit(c => c.copy(name = f(c.name)))
  def mapChiselAnnos(f: Seq[ChiselAnnotation] => Seq[ChiselAnnotation]): ElaboratedDesign = {
    mapCircuit(c => c.copy(annotations = f(c.annotations)))
  }
  def mapComponent(f: Seq[cir.Component] => Seq[cir.Component]): ElaboratedDesign = {
    mapCircuit(c => c.copy(components = f(c.components)))
  }
}
```

With #2274 `cir.Component` is no longer available, and we also (in mapComponent use) match components against `cir.DefModule` in which we retrieve ports (`Port`).

As our usage of `Component` and `DefModule` remains quite limited, I guess we could achieve the same results with the content of `chisel3.stage.DesignAnnotation` but I would greatly appreciate if you could add a deprecation phase for `chisel3.internal.firrtl.Component` and `chisel3.internal.firrtl.DefModule` just to give us some flexibility (ie upgrading to 3.5 without errors and then upgrade our code).

**Please tell us about your environment:**
    - version: migrating from `3.5.0-RC1` to `master`

**What is the use case for changing the behavior?**
Please do not break (even evil) user code in the middle of a release process without deprecation phase.

ekiwi, 2021-12-17:
Not commenting on whether this should be reverted or not. But just as a possible work around: You can always make a package inside of your project that matches the one that contains the private classes. I have done that in the past for multiple such "hacks": 
- https://github.com/ekiwi/paso/blob/main/src/chisel3/hacks/ElaborateObserver.scala
- https://github.com/ucb-bar/maltese-smt/blob/main/src/firrtl/backends/experimental/smt/ExpressionConverter.scala

jackkoenig, 2021-12-17:
mea culpa on this one 🙂 

I figured we could sneak it in and if anyone complained we could revert and then do a more formal deprecation schedule. I assumed no one would notice till it was released in 3.5.0-RC2, but thank you @johnsbrew for your diligence in reporting this even sooner! I'm going to revert and then look into a deprecation schedule for 3.5.

Note that I would not encourage @ekiwi's approach here because part of why I want to make these APIs private is so we can change them arbitrarily.

closed 2021-12-18 by mergify[bot]

johnsbrew, 2021-12-18:
Thank you very much, sorry for being the annoying user 😬

jackkoenig, 2021-12-18:
The revert made it in to `3.5.0-RC2` so you code should not be broken there. I didn't want to delay to add the deprecations, but you can expect them before `3.5.0` 🙂