chipsalliance/chisel #2325: 3.5.0-RC2 breaks all tests from the Chisel book

schoeberl, 2022-01-03:
**Type of issue**: bug report

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

Switch to RC2 in https://github.com/schoeberl/chisel-book.

**What is the current behavior?**

ScalaTest 3.2 changes classes and therefore breaks ScalaTest 3.1 (ChiselTest) code. See https://github.com/ucb-bar/chiseltest/issues/368

**Solution**

Change back to 3.1.4 and provide a warning that this may change in the future.

schoeberl, 2022-01-03:
Needs probably to be done on all Chisel libraries.

ekiwi, 2022-01-03:
This is most likely the fault of `iotesters`: https://github.com/ucb-bar/chiseltest/issues/368#issuecomment-1004311026

closed 2022-01-03 by ekiwi