chipsalliance/chisel #2388: Hierarchy API does not support @public on `val`s in constructors

debs-sifive, 2022-02-02:
**Type of issue**: feature request

**Impact**: API addition (no impact on existing code)

**Development Phase**: request

**Other information**
 

This is pretty easy to work around, but it might be nice to have support for this.

https://scastie.scala-lang.org/debs-sifive/zAVcJfvxTnGqhKLbTa7sUw/51

This does not work:
```scala
@instantiable
class PassThrough(@public val hello: String) extends Module {
  ...
```

Workaround:
```scala
@instantiable
class PassThrough(val hello: String) extends Module {
  @public val hello = hello
  ...
```

**Please tell us about your environment:** See Scastie.

**What is the use case for changing the behavior?** Less typing.

mwachs5, 2022-02-02:
I think this is the documented behavior...

debs-sifive, 2022-02-02:
@mwachs5 do you know where that's documented?

mwachs5, 2022-02-02:
See the hierarchy cookbook, search for `@public val clock`

mwachs5, 2022-02-02:
https://github.com/chipsalliance/chisel3/blob/master/docs/src/cookbooks/hierarchy.md

_To mark a superclass's member as @public, use the following pattern (shown with val clock)._
```
import chisel3._
import chisel3.experimental.hierarchy.{instantiable, public}

@instantiable
class MyModule extends Module {
  @public val clock = clock
}
```

debs-sifive, 2022-02-02:
Oh gotcha, yes the workaround is pretty much to do what's normal. I just think it's kind of confusing/verbose to have to make another `val` and `@public` the second one.