chipsalliance/chisel #2408: `assert` emits `$fatal` before error messages in generated SystemVerilog

CircuitCoder, 2022-02-10:
**Type of issue**: bug report
**Impact**: API modification
**Development Phase**: request

**Other information**
As per the implementation of `assert`
https://github.com/chipsalliance/chisel3/blob/024847d75079a125e5946e9dcf2ed9c14d2db730/core/src/main/scala/chisel3/VerificationStatement.scala#L94-L97
A verification statement is generated before the failure message, which gets turned into a `$fatal` statement by firrtl. This makes the failure message never visible in simulations.

**steps to reproduce the problem**
Call `assert` in any elaborated code.

**What is the current behavior?**
`$fatal` generated before `$fwrite`

**What is the expected behavior?**
`$fwrite` generated before `$fatal`, or at least have a switch for changing the order.

**Please tell us about your environment:**
Windows 11 WSL2, ArchLinux

**What is the use case for changing the behavior?**
Better debug experience using external simulator.

ekiwi, 2022-02-10:
I can see how this will be a problem. If you want to have a quick work around, you could try to use the SystemVerilog emitter which will generate a real `assert` instead of a `$fatal`, maybe that helps.
The flag is `-E sverilog`.

CircuitCoder, 2022-02-11:
@ekiwi I just tried out the SystemVerilog output, and it does generates `assert` calls. But all assertions are generated in a separated `always` block, and verilator somehow decides the assertions should run first, so the failure messages is still not visible. Manually moving the assertions to the same `always` block of the `$fwrite`s produces the expected output.

Actually I thought I was using the systemverilog emitter (with the -E sverilog), and it does emit a `.sv` file, but since I used `ChiselState.emitVerilog` I suspect it forces firrtl to use the verilog emitter? So the title of this issue is probably a little bit misleading. I'm going to change that.

jerryz123, 2022-04-14:
Is there any activity on this? This is a significant nuisance on 3.5.0 and newer.

mwachs5, 2022-04-14:
Is this actually a chisel3 issue, or is this related to firrtl's translation from the .fir to .v?

ekiwi, 2022-04-14:
## Integrating the Message with the Verification Statement

> It's complicated... There are two interacting things:
> 
>     1. Chisel emits these asserts as an `assert`/`printf` pattern (with the assert before the printf). Chisel should really be emitting this as just an assert with a message.
> 
> It seems cleanest to fix (1).

Unfortunately (1) won't work without some significant work on the firrtl compilers. You would have to extend the assert and assume statements to support format strings and potentially duplicate some code for handling print statements. I think this would be a nice solution, just want to point out that there would be some work involved.

## Assertion Message Semantics

The Verilog that the C++ firrtl compiler generates would work well when there is, however what would happen if there are two assertions that fail in the same cycle? What if they are in different modules? Are we giving any guarantees in that case?
Would it be OK for the Scala firrlt compiler to move all assert/stop statements below all printf statements? I.e. we would guarantee that printfs inside a module happen in order, but assertions will always happen after all printfs (inside a module).

## Fixing this Issue

I have a short fix which should bring things back to the state we had in `3.4`: https://github.com/chipsalliance/chisel3/pull/2484

jerryz123, 2022-04-19:
Thanks for the fast fix.

closed 2022-04-19 by ekiwi