chipsalliance/chisel #2409: Definition/Instance and doNotDedup Interaction

seldridge, 2022-02-11:
Presently, any use of `NoDedupAnnotation` or it's Chisel-level API, `doNotDedup`, will be silently ignored by a Chisel compilation.

Consider:

```scala
import chisel3._
import chisel3.experimental.hierarchy.{
  instantiable,
  public,
  Definition,
  Instance
}
import chisel3.stage.ChiselStage

@instantiable
class Bar extends Module {
  @public val in = IO(Input(Bool()))
  @public val out = IO(Output(Bool()))

  out := in

  chisel3.experimental.doNotDedup(this) // <-- This doesn't matter in D/I land
}

class Foo extends Module {
  val in = IO(Input(Bool()))
  val out = IO(Output(Bool()))

  private val barDef = Definition(new Bar)

  val bar1 = Instance(barDef)
  val bar2 = Instance(barDef)
  

  bar1.in := in
  bar2.in := in
  out := bar1.out ^ bar2.out
}
```

This is a main module, `Foo`, that instantiates two copies of the same definition of `Bar`. However, `Bar` has internally declared itself as `doNotDedup`.  Then end result of this elaboration (and compilation with a FIRRTL compiler) is that `Bar` is deduplicated.

I'm not sure how to handle this... There are only a limited number of places where Chisel uses this API (including a handful of SiFive-internal utilities as well as) `InlineInstances` and `BoringUtils`.

What I think is going on here is that `doNotDedup` is a relic of when annotations were limited to being `Named` where the specificity of `Target` to specify a path to a _specific_ instance was not present. In that world, annotations used to block deduplication because specificity was lost by allowing un-annotated and annotated things to merge. We eventually relaxed that, but then we needed a way to keep legacy annotations working (whose transforms required that they were operating on unique instances).

What we could probably do here is to work towards full removal of `Named` (migrating to `Target` for annotations which don't currently use it) and updating downstream transforms to understand non-local `Target`s would enable removing this API and allowing people to D/I their designs however they want. Alternatively, FIRRTL compilers may need to change their interpretation of `NoDedupAnnotation` from "do not deduplicate me" to "make sure I am unique".

mwachs5, 2022-02-11:
> Then end result of this elaboration (and compilation with a FIRRTL compiler) is that Bar is deduplicated.

Just to clarify, the end result is that there is only one `Bar.v`. 
Are there valid use cases for using D/I and doNotDeuplicate at the same time? If that's what you mean, why not explicitly make two different Definitions?

mwachs5, 2022-02-11:
> Alternatively, FIRRTL compilers may need to change their interpretation of NoDedupAnnotation from "do not deduplicate me" to "make sure I am unique".

seldridge, 2022-02-11:
> Are there valid use cases for using D/I and doNotDeuplicate at the same time?

It's usually a pattern of:

1. I want a Chisel API that is backed by a FIRRTL transform
2. I don't want to have to deal with non-unique instances in the FIRRTL transform
3. I drop a `NoDedupAnnotation` down to guarantee this.

There are newer, better ways of doing this using `EliminateTargetPaths` to manipulate the hierarchy to guarantee that instances with specific annotations are unique before the transform runs. However, we have legacy Scala FIRRTL Compiler transforms (and internal stuff) that doesn't use this.

> If that's what you mean, why not explicitly make two different Definitions?

That annoying won't work because the `NoDedupAnnotation` is trying to guarantee that this instance is unique regardless of what happens above it. Requiring multiple definitions would mean that a user needs to have complete knowledge of the sub-hierarchy of a definition before instantiating it.

(This is all beside the point. The real issue is that we need to fix transform-backed APIs that use `NoDedupAnnotation` when they should use `EliminateTargetPaths` or they should be updated to work on instance annotations.)

> That seems reasonable... then the FIRRTL action here would be to explicitly make two Bar.v and Bar_1.v because of the annotation?

The FIRRTL compiler requirement is pedantically that any transform in a FIRRTL compiler will never see more than one instance of `Bar`. That's what `NoDedupAnnotation` means right now. Put differently, `NoDedupAnnotation` is an API contract between how Chisel generates FIRRTL IR and what a FIRRTL compiler can do to a circuit when deduplicating.