chipsalliance/chisel #2413: Isn't .U syntaxt supposed to only take .W arguments?

mwachs5, 2022-02-14:
**Type of issue**: bug report

**Impact**: API modification

**Development Phase**: request 

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

Put this in a `Module`:

```  
val bar = 5
val foo = bar.U(7)
```

Here is a simple Scastie (some commented out code that also all works unexpectedly)
https://scastie.scala-lang.org/nqEQGHEUTziPKNj7NUirBA

**What is the current behavior?**

This code compiles without error

**What is the expected behavior?**

I thought it should be an error to have no `.W` suffix on the arguments inside the `.U`, since those are not supposed to directly take integer arguments

**Please tell us about your environment:**

    - version: 3.5.1
    
**What is the use case for changing the behavior?**

Hard-to-catch unintended user errors

jackkoenig, 2022-02-14:
This is something we've discussed in the past, I attempted to fix in https://github.com/chipsalliance/chisel3/tree/deprecate-literal-bit-extract and @albert-magyar had a PR proposed to fix via ScalaFix and linting: https://github.com/chipsalliance/chisel3/pull/1449

mwachs5, 2022-02-14:
>  I attempted to fix in https://github.com/chipsalliance/chisel3/tree/deprecate-literal-bit-extract

What was the issue with the attempt?

jackkoenig, 2022-04-26:
> What was the issue with the attempt?

I just ran out of steam and was struggling getting all of the cases working. I think we should redo that branch and just do `.U` and `.S`. We also need to make sure those methods don't have parentheses. We can make the suggestion for users who do want to extract immediately to call `.extract` which is a new method we can add that `.apply` is an alias for.

PR #2534 (2022-06-02): Deprecate implicit .U() and .S() syntax for literal bit extracts
Fixes #2413: It is currently possible to pass an integer to the literal .U/.S syntax as a way of immediately bit extracting on a literal value, but this poses confusion and ambiguity as users usually intend to pass a Width.

### Contributor Checklist

- [ ] Did you add Scaladoc to every public function/method?
- [ ] Did you add at least one test demonstrating the PR?
- [x] Did you delete any extraneous printlns/debugging code?
- [x] Did you specify the type of improvement?
- [ ] Did you add appropriate documentation in `docs/src` [... cut]
closed 2022-06-02 by jackkoenig