chipsalliance/chisel #2451: BlackBox Crashes if io <: Record is Absent

seldridge, 2022-03-17:
BlackBox will give a glorious `None.get` crash if you forget to provide an `io <: Record` member.

```scala
class BlackBoxModule extends BlackBox {
  val io = IO(Output(UInt(1.W)))
}

class Example extends Module {
  val foo = Module(new BlackBoxModule)
}

chisel3.stage.ChiselStage.emitVerilog(new Example)
```

Produces:

```
java.util.NoSuchElementException: None.get
	at ... ()
	at Playground$Example$$anonfun$2.apply(main.scala:10)
	at Playground$Example$$anonfun$2.apply(main.scala:10)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
	at Playground$Example.<init>(main.scala:10)
	at Playground$$anonfun$3.apply(main.scala:13)
	at Playground$$anonfun$3.apply(main.scala:13)
	at ... ()
	at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
```

The error here should, at worst, be better and, at best, consider being a compile-time error where `io` is abstract for `BlackBox` (even if it isn't for module).

Alternatively, `BlackBox` could be made more like `ExtModule` and not require everything to be funneled through `io`.

ekiwi, 2022-03-17:
> Alternatively, `BlackBox` could be made more like `ExtModule` and not require everything to be funneled through `io`.

I feel like the long-term goal is to deprecate `BlackBox` in favor of `ExtModule`. But I think I remember that there were some features missing for `ExtModule`. If we are going to deprecate `BlackBox` (which would be in keeping with the philosophy of providing one way to do things) then adding a better error message should be enough.

jackkoenig, 2022-04-04:
Discussed a bit in Dev today. Indeed deprecating `BlackBox` in favor of `ExtModule` is the plan, we just need to finish `FlatIO`. Implementing `FlatIO` merely requires 1 minor enhancement to `DataView`--namely that we need to be able to dynamically index Vecs that are part of a larger view. See this related issue that fixed a similar problem but now the full thing: https://github.com/chipsalliance/chisel3/pull/2298

closed 2022-04-25 by jackkoenig