chipsalliance/chisel #2453: Fix verilog vs chisel doc remdering

debs-sifive, 2022-03-21:
**Type of issue**: bug report | documentation

**Impact**: no functional change

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

visit https://www.chisel-lang.org/chisel3/docs/cookbooks/verilog-vs-chisel

https://github.com/chipsalliance/chisel3/pull/2323#issuecomment-1074349506

**What is the current behavior?**

https://www.chisel-lang.org/chisel3/docs/cookbooks/verilog-vs-chisel does not render HTML properly.

**What is the expected behavior?**

* HTML code should not show up in the generated doc
* code blocks should work

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

Shorla, 2022-03-30:
@mwachs5 can I make more research and try working on this?

debs-sifive, 2022-03-30:
@Shorla absolutely! that would be super helpful :)

pmusa, 2022-06-07:
Hi @debs-sifive, is there a way to test the website results locally? The IntelliJ preview is not showing the issue.

Shorla, 2022-06-23:
@mwachs5 @debs-sifive @pmusa, after much deliberation with of making Markdown and Html render with almost no positive result. I am suggesting we look for a different for way to make the difference visible?